hneemann/Digital #327: Auto-width of constants, pulls and supplies

SmallRoomLabs, 2019-09-08:
One thing that annoys me a bit is that I have to set the width of the Constants, Pull-up/downs and Supplies to match whatever wire I attach them to.

Since those devices *must* have the same widths as the attached wire  I see no real reason for me (a human) to spend time and annoyance for something the computer can do for me automagically without fail.

I might have missed some feature/function in Digital where it's possible or even desirable to have a one of the above components with a differing width than the wire they connect to - in that case this enhancement-request can be put in /dev/null ;-)

hneemann, 2019-09-09:
This is difficult to add because when drawing the circuit there is only a pile of blue lines and boxes with red and blue dots. Inconsistencies are only found when starting the circuit.
But maybe one can build something, which makes the fixing easier.
Please try the latest pre-release and let me know if this helps. But to be honest, I tend not to add this feature because it adds a lot of complexity and brings little use.

SmallRoomLabs, 2019-09-09:
Was this comment in relation to the auto-widths or for different coloring of single/multiple-wire nets?

Anyhow...  I definitely don't expect you to actually implement all of my enhancement suggestions - only those which makes sense to you and the current codebase.

SmallRoomLabs, 2019-09-09:
After re-reading your comment I realize that it was about the auto-width....  But unless you have implemented it all in a completely different way than I would I recon that it would be possible to auto-adjust the width of the pulls&constants in the "inconsistency checker/error message" function at the first run.

SmallRoomLabs, 2019-09-09:
The auto-set-width in v0.23-65 is really nice!  Not only does it handle the powers and pulls but also handles other cases with real signals.  👍 

[image]

closed 2019-10-25 by hneemann