hneemann/Digital #328: Is there a way to find tunnels/nets?

SmallRoomLabs, 2019-09-09:
Can I search/find all tunnels/wires/nets with a particular name somehow?

Like if I have an output named AccZero , and I want to know if and where it is tunneled to - because maybe I don't even use that signal elsewhere anymore so I can safely remove the circuit that generates the AccZero signal.

To know if it's even used at all is easy, I can just delete the signal and try to run and see if I get an error.  But to be able to see where it's used would be convenient sometimes.

hneemann, 2019-09-09:
You can activate wire tool tips in the settings, i.e. when the mouse moves over a wire, the wire is highlighted. And this applies to the whole net, across all tunnels.

SmallRoomLabs, 2019-09-09:
Mmm, yes. That works kinda ok.  The highlighting of the net (changing from blue to a lighter shade of blue) is a bit hard to see on a larger schematic.  I'm happy with this for the time being. ^__^

closed 2019-09-09 by SmallRoomLabs