hneemann/Digital #329: Differentiation between single & multiple wire nets

SmallRoomLabs, 2019-09-09:
I use Eagle a lot for making PCBs and there I can draw (and name) a bus consisting of many nets and then "peel off" individual wires from it.  Here a splitter is used for the same thing which is ok.

But I miss the distinct representation of a bus and a single wire. A bus is  thick and blue and a wire is thin and green.  This makes it immediately obvious how the major data-paths are distributed among the schematics, and what that is just random control signals for the logic.

[image]

hneemann, 2019-09-10:
See also #21

hneemann, 2019-09-10:
There is a flag in the settings which allows to show the number of wires represented by a bus.
This is the solution i came up with in #21.

SmallRoomLabs, 2019-09-10:
I've got the number-of-wires -flag on already, but it's not visible when not running and the font is so small it's unreadable unless I've zoomed in a lot.v So that is not an ideal solution, but still better than nothing. ;-)

Since this already have been discussed and thought about I'll accept that previous conclusion.  So I'll close off this issue.   

But I'll keep on opening issues for things that I realize that I wish for while using Digital....

closed 2019-09-10 by SmallRoomLabs