hneemann/Digital #330: An unexpected failure in the NMOS XNOR example

keller91711, 2019-09-10:
Could you please try the example nmos\nmos-xnor.dig 
Try A = B = 1, then set B = 0. We sometimes get an error dialog:

Error calculating a step
More than one output is active on a wire, causing a short circuit.

It seems like the circuit should work, but maybe there is a race issue?

hneemann, 2019-09-10:
There is no example file nmos\nmos-xnor.dig, but there is a file nmos\xnor.dig.
Do you mean that file? I Have testes it heavily and it seems to be ok.
Could you please confirm that the original file nmos\xnor.dig causes the problems or, if not, attach the problematic file? 

This is the circuit in the original nmos\xnor.dig:
[image]

keller91711, 2019-09-10:
Sorry about the naming mistake. That is the circuit. I can't seem to
reproduce it on the version of Digital that I just downloaded, which is
Build git-Revision: v0.23, Build Time: 2019-08-23 11:08

 I'm attaching the screenshot from before. Is it not the identical circuit?
Two students first called it to my attention. I then shot this from
WIndows. I was not able to reproduce it on the Mac version, but one student
said he was.

I hope I'm not going crazy.

[image: image.png]

hneemann, 2019-09-10:
I think I figured out what happened. Did you use the circuit that comes with Digital, or did the students build it themselves?

hneemann, 2019-09-10:
Due to limitations in the simulation engine it is not possible to simulate all cmos/nmos circuits. This limitation is caused by the fact, that Digital does not simulate the residual FET resistance if the FET is conductive. (See #30, #22, #29).
To overcome this limitation the FATs can be used in a unidirectional mode. And this mode is also necessary to simulate the circuit mentioned above. And this means that, if the option "Unidirectional" is not set in both FETs, the circuit will not work and behave as observed.

keller91711, 2019-09-10:
Yes, that indeed explains it. The students constructed the circuit from an
image, but would not have known to set Unidirectional, nor did I.
A lingering question though is whether the circuit would work in reality.
Is there a reference on this?
Thanks.
Bob

hneemann, 2019-09-10:
Do you have any doubts? I don't have a reference at hand, but you can simulate this circuit with a more capable analog simulator like Spice, which also takes into account the residual resistance of the FET's. 
And such a simulation works pretty well:
[image]

keller91711, 2019-09-10:
I will take your word for it. I don't do much electronics anymore, but your
comment is reassuring. That circuit seems super elegant.
Thanks.