hneemann/Digital #335: Bookmarks for screen position

SmallRoomLabs, 2019-09-13:
You know what would save (at least) me a ton of time while debugging my design?

Bookmarks.  Being able to press like shift-control-1 to capture the current scrollbar-locations and the zoom level into the first bookmark-slot.  And then just press control-1 to instantly recall that position.  Maybe up to 9 slots.

That would be an awesome timesaver for me. As is it now I feel I'm spending more time scrolling and zooming into 2-3 different places that anything else while debugging. Lookin at my front panel, heading over to the main datapaths for the cpu, and then up to the controller logic to see current microcode step and exactly which control signals begin asserted now and what will happen in the next step - and then back down to the datapaths again to verify.

hneemann, 2019-09-14:
That might be a good idea. To be honest, I never work with circuits that big. I tend to break them down into smaller partial circuits, which keeps the circuits small.
I will think about it.

closed 2019-09-14 by hneemann

hneemann, 2019-09-14:
Now its possible to store a view position with CTRL+[n] (COMMAND+[n] on MacOS) and to restore the position with [n].

SmallRoomLabs, 2019-09-14:
Holy cr*p!  You are fast!  I'll try it out straight away...

SmallRoomLabs, 2019-09-14:
The setting and recalling of the bookmarked positions works fine.  But they seem to not be saved in the .dig file so they need to be setup for each load.

SmallRoomLabs, 2019-09-14:
Ollrighty!  Now it is perfect...  A definite time-saver!   Thanks.

SmallRoomLabs, 2019-09-15:
Hmm... It seems like I spoke too soon again. ;-)

Zooming affects the current recalled favorite and gets auto-stored in it.

SmallRoomLabs, 2019-09-16:
Okkidokie, I'll give it a spin and see!

BTW, a link in the main README.md to the pre-release download server would be nice. It takes me a while to find it every time I want to download from there.,,,,

SmallRoomLabs, 2019-09-20:
After a few evenings of using Digital I can confirm that this  bookmark/favorite function actuallys saves me a ton of time.  In the design I'm doing now I have 5 different bookmarks that I flip around to both while debugging/running and (to some extent) while drawing.

commit 27eee49e (2019-09-14): adds view favorites, closes #335