hneemann/Digital #337: Wire splitter issue

kierangilliam, 2019-09-20:
I am finding that I am running into this same issue across Digital. 

"Error creating the circuit. No output connected to wire... The state of the wire is undefined."

The wire splitter output is connected directly to the a circuit's inputs but that does not register.

[image]
[image]
[image]

kierangilliam, 2019-09-20:
Adding another example.

[image]

closed 2019-09-20 by kierangilliam

Usama148-hub, 2022-04-03:
same error I am facing

hneemann, 2022-04-03:
@Usama148-hub 
Digital splitters are directional. It could be that you are using the splitter in the wrong direction.

prlaba, 2022-04-04:
Splitters are cool!  But they could probably use better explanations in your User Guide.

Explaining more fully how the Fanout attribute works, with examples, would also be helpful.  When set up correctly, you only have to expose the signals you want, without having a bunch of unconnected leads coming out of or into a splitter.