hneemann/Digital #342: Simulating clock rising trigger: What is wrong with my circuit?

trungdq88, 2019-09-23:
I'm trying to simulate the clock rising trigger. It works for the built-in D Flip Flop but doesn't work for the one I built with NOR gates. I attached a GIF below.

[image]

I really appreciate someone could help me point out what's wrong, thank you.

hneemann, 2019-09-23:
The principle is correct, but the trigger pulse is too short.
If the circuit is started in "single gate mode", you can observe in detail what happens.
To fix this, the trigger pulse must be extended to at least two gate propagation times.
Instead of using one NOT-gate you could use e.g. three.
The built-in D-flipflop works because it only requires one gate propagation time to switch. The NOR flipflop requires two.

trungdq88, 2019-09-23:
Very nice! Thank you!

closed 2019-09-23 by trungdq88