hneemann/Digital #343: Undo implementation?

SmallRoomLabs, 2019-09-24:
Just out of curiosity I'm a bit interested in how the implementation of the undo is done in Digital.

My schematics is large enough now that a single undo takes almost 5 seconds, but all other operations is as snappy and nice as always.

Without having looked at the code I almost guess that the undo is actually clearing everything and then "replaying" each step made until the second-to-last.

closed 2019-09-24 by SmallRoomLabs

hneemann, 2019-09-24:
Maybe I should keep a snapshot with about 10 changes in the past. Then there would only be a noticeable delay if more than 10 undo's were executed in a row. I have to think about it.

SmallRoomLabs, 2019-09-24:
I'd guess 99% of my undos are just a single undo to get rid of an unwanted segment of wire (happens quite a lot) or when I move something and it ends up overlapping something else.

So for me 2-3 speedy undos in a row would absolutely be enough.  

Gotta think about the memory usage to save all those snapshots ;-)  But maybe they are not that large even for a big design in comparison to available memory on the computer.  (I mostly work with memory-limited devices so I might be a bit biased when it comes to spending memory)

But all in all, I don't see this as a huge problem now when I know that I can just reload the design and get full undo-speed again.  Just mention it in the readme.

hneemann, 2019-09-24:
I would only keep a single additional snapshot which represents the state 10 modifications in the past.
If you do more than ten undo's you have to start at the beginning. But the next undo is again fast. Only when again 10 undo's are carried out it takes a while again. 
So there is no big memory consumption. Currently two states are held, then it would be three.