hneemann/Digital #344: Component "Counter" caps at 64 bits but only works up to 63 bits

SmallRoomLabs, 2019-09-24:
At 64 bits it's not counting at all, just outputting zero.

With "caps" I mean that if I enter the number of bits as 100 it gets (silently) auto-adjusted down to 64 - which in this case doesn't work.

closed 2019-09-24 by hneemann

SmallRoomLabs, 2019-09-24:
I was actually testing this on the simple non-preset counter - so you need to fix that one as well ;-)

hneemann, 2019-09-24:
I did a "1L <<" full text search to find all incidents that have made it into the code since the last such search. And you are right, both counters had the problem.
I've also been thinking about writing a test to test the overflow. Unfortunately it takes 100 years for an overflow to occur with a 6MHz clock. :smile:

SmallRoomLabs, 2019-09-24:
That might be a bit too long - one have to rent a huge cluster of EC2 instances to run the tests then. ;) 

And I thought that my PDP-8 simulation was a bit slow compared to the original machine. It runs at about 1/60th of the real speed.  So a diagnostics that normally sends a beep every 5 seconds (X thousands of successful iterations) takes 6-7 minutes in the simulation.  But there's a test that takes 35 minutes on a real machine to loop thru once. I don't even dare to run that in the simulation - that would be like 35-40 hours. 

Another question:  I'm trying to compile the latest source on the master branch with `mvn install -Dmaven.test.skip=true`. But it fails with some coverage error that I've got no idea of what it really is and how to rectify.

[image]

hneemann, 2019-09-24:
could you please try

`mvn clean install -Dmaven.test.skip=true`

SmallRoomLabs, 2019-09-24:
Ah yes - that worked much better.  Thanks. :+1:

commit 7005add0 (2019-09-24): fixes a bug in the counter, closes #344