hneemann/Digital #347: JAR File

vcokltfre, 2019-09-26:
This may be me being stupid but I can't find the JAR file to start (Linux) , and when I downloaded it on Windows the EXE couldn't find it either.

SmallRoomLabs, 2019-09-26:
Download the Digital.zip from the "Releases page".  There you have the proper exe's, jars and scripts in the root folder of the zip.

[image]

SmallRoomLabs, 2019-09-26:
https://github.com/hneemann/Digital/releases

hneemann, 2019-09-26:
@vcokltfre 
The git repository does not contain the JAR file. It only contains the source code.
To get a executeable jar file you have to download a [release](https://github.com/hneemann/Digital/releases/latest) as pointed out by SmallRoomLabs or you have to build the binary yourself from the sources.

closed 2019-09-26 by hneemann

hneemann, 2019-09-29:
See also #321