hneemann/Digital #348: Memory usage, loading and saving

keller91711, 2019-09-27:
Does "program memory" imply read-only, or can we have a stored-program memory to which we can also store data?
What is the format of loadable memory?
Is there any way to make the tiny spreadsheet table larger?
Is there a way it could be loaded from another spreadsheet or, say, CSV file?
It is useful, but retyping the contents every run is a pain.

SmallRoomLabs, 2019-09-27:
I'm using the EEPROM-component with the "Program memory" checked as the memory for coda/data in my PDP-8 design I'm currently working on. 

The contents of the memory is then automatically be (re-)loaded from a plain binary file at every run of the simulation, this is specified in the _Edit | Circuit specific settings_

[image]

[image]

hneemann, 2019-09-27:
If memory is marked as "program memory", this has no effect when the simulation is running.
When initializing the simulation model, however, this memory can be initialized with a file (see above). Therefore RAM can also be used as program memory. The TCP/IP API used by a Assembly IDE also uses this flag to load a hexfile into this memory.

If the file extension ".hex" is used, the file must correspond to the old Logisim file format or it must be an Intel hex file.
With any other file extension, the contents of the file is read directly into the memory as a binary file.
CSV is not supported. But the old Logisim file format is very easy to create. Here is an example:

```
v2.0 raw
200080
4004c0
4044c0
4008c0
4048c0
400cc0
4010c0
4014c0
```
The first line is a fixed header, followed by on hex value per line.

It is also possible to divide the address space and, for example, use ROM for one half of the address space and RAM for the other half:

[image]

keller91711, 2019-09-27:
This was helpful. Thanks.

closed 2019-09-27 by hneemann