hneemann/Digital #349: Consistent display of values in measurements

SmallRoomLabs, 2019-09-27:
In the CSV file created when saving the measurements data the values are typically hex - at least from most components.

As long as the value contains a number A..F there are no 0x-prefix on the data, but when there's just plain digits there's ambiguity in regards to the radix the 0x-prefix is prepended to the number.

I'd prefer that the output would we consistent with a 0x at all times if a column is in hex to easy both manual reading of the data and parsing it with software.

[image]

hneemann, 2019-09-27:
Please try the latest [pre-release](https://infdigital.dhbw-mosbach.de/),

SmallRoomLabs, 2019-09-27:
v23-95 is better, but still displays single digits without any prefix (as in the AC column in the screenshot)

Also, all buttons are always in the measurements table - and there's no checkbox to turn measurements on/off in the buttons dialog.

[image]

hneemann, 2019-09-27:
Which format is selected for the AC signal?

SmallRoomLabs, 2019-09-27:
It's a D-FlipFlop so the format can't be set.

I now tried adding three probes on the AC-bus instead of measuring on the component itself.

The hex and octal format probes looks ok. But the one with default shows single digit "decimal" for values less than 10 and 0x-prefixed hex for larger values.

hneemann, 2019-09-27:
The default format is intended to be read by humans.
A value smaller than 16 requires no suffix: The values 0-9 are equal in hex and decimal and the values A-F are clearly hex. If a larger value is displayed and the hex value contains a letter, no prefix is required: The number must be hex. If the number contains no letter, a prefix is introduced, otherwise '10' and '0x10' would look the same.
This is why the value are more difficult to parse by a program. I now replace the default format by the hex format in the csv export if more then 3 bits are used. Please try the letest pre release again.

SmallRoomLabs, 2019-09-27:
I understand the reasoning and it's fully valid.  The root cause for me was that I couldn't set the default radix on the flip-flop and it turned out a bit strange in my case.

I tried the 23-97 and the new solution seems to work very well.  Easy to read for smaller numbers and when larer numbers are possible it has a consistent look.

SmallRoomLabs, 2019-09-27:
BTW - is there a way for you to control how the dialog boxes acts? 

 As it is now both the "save" dialog and the next "overwrite" dialog both pops up *under* the  measurement window.  This makes me sit there for a while thinking that the software lags/is slow until I remember that I have to move the measurement window to see the save-dialog.

I probably learn this and do it automatically in a couple of hours  ;-)

hneemann, 2019-09-27:
That's interesting! This does not happen on my Debian machine.

hneemann, 2019-09-30:
Today I could test the problem with the window order on a Win7 machine and it works just fine.