hneemann/Digital #350: Just an observation on the GUI

SmallRoomLabs, 2019-09-28:
While running a long diagnostics run on my cpu-design the screen goes low-res. Moving around to different locations in the design (while still running) doesn't make it hi-res and prettu again. 

Not until I stop the sim (and I think I have to scroll as well) does the screen go hi-res again.

This is not a problem at all, but it is a kinda funny effect.

**How it normally looks / hi-res:**

[image]

**How it looks after 10 minutes running / lo-res:**

[image]

hneemann, 2019-09-28:
This behaviour is intended. If the screen refresh takes too long (longer than 500ms) "anti alias" will be switched off to keep the surface reactive. If the first time the screen refresh time is less than 50ms, "anti alias" will be turned on again.

SmallRoomLabs, 2019-09-28:
Good to know it's not some strange artifact caused by bad video drivers or something like that...

closed 2019-09-28 by SmallRoomLabs