hneemann/Digital #351: Auto-generate test data from current design?

SmallRoomLabs, 2019-09-28:
A part of my design is a big mess of combinatorial logic with a lot of gates with inverted logic and so. As I sat down to try to clean it up a bit and make it more "actually buildable" it occurred to me that it might be good to learn how to use your little green test-box.

After copying my messy part into a separate drawing and attaching proper input- and output-boxes to the relevant points I could use F9 to get a truth-table and a bunch of equations but it seems like I can't copy or save the truth table into a format that the test-box can accept.  Nor can the test-box autogenerate the tests from the current design.

Wouldn't it be incredibly convenient to have the Analyze and Test compatible with eachother so I don't have to write a 10-input 4-output testcase data file by hand?  ^__^

hneemann, 2019-09-28:
I'm not sure!
The genereted test would not be a test case, because it would always be green.
It wouldn't test if the circuit works the way it should, but just that it is the way it is.
And if you want to create an elegant circuit, you can simply auto create a clean circuit from the truth table.

SmallRoomLabs, 2019-09-28:
That is true. It would be more of a reference-case.  Still, it could be useful to be able to export the truth-table in a cvs and/or textual format.

But I'm not sure I would call the autogenerated circuit elegnt. ;-)  From an aestetics point of view the autogenerated are beautiful in a mechanical and strict way.

But compare these two.  One conveys the workings of the design more than the other.  That is beautiful to me...

[image]
[image]

hneemann, 2019-09-28:
It was so easy. No reason not to implement it.

closed 2019-09-28 by hneemann

SmallRoomLabs, 2019-09-28:
Heh....  I like when the enhancements are easy to implement ;-)

SmallRoomLabs, 2019-09-28:
I found a small problem.  

When Analyzing a circuit with multi-bit inputs or outputs the truth table appends a _0 , _1 and so on onto the name of the input/output.  But the Tester is not happy with those names.  I guess multibitters should be entered as the base-name with a value varying from 0 to 2^n in the Tester.

It's easy enough to just split up the multibitters to singlets manually and then analyze, so it's not really a big deal.

kEviN1H, 2019-12-09:
> That is true. It would be more of a reference-case. Still, it could be useful to be able to export the truth-table in a cvs and/or textual format.
>
> But I'm not sure I would call the autogenerated [...]

howto export  the truth-table to a cvs and/or textual format

hneemann, 2019-12-09:
@kEviN1H 
Use the latest [pre release](https://infdigital.dhbw-mosbach.de/). In the table window, use _File->Export Test Case_ or _File->Export LaTeX_ to copy a textual representation to the clipboard.
_File->Export to Hex_ exports the result columns to a simple hex file representation which can be imported into a ROM to create a look-up-table in a circuit.

kEviN1H, 2019-12-09:
OK, Thanks a lot!!!!

commit 8a757445 (2019-09-28): In the truth table it is possible to create a test case description, closes #351