hneemann/Digital #353: Running Test case from command line

d-teif, 2019-09-30:
Hello,

I would like to now if there is a way to run **test case**  from command line and obtain status. This will be helpful for batch evaluation program. 

Thank in advance 
Didier

hneemann, 2019-09-30:
This is not yet possible. But it sounds like an interesting feature.
Can you explain your use case in more detail?

d-teif, 2019-09-30:
It's for automatic students circuits evaluation.  
When they return their files I would like to run automatly  test  from command line with script.

 **Option 1 :** Testing with student test case code include in circuit (not always correct or complete)
For example java -jar Digital foo.dig - testCaseIntegrate return 1 if OK, 0 if not

**Option 2 :** Testing with my test case code (correct and complete)
For example java -jar Digital foo.dig - testCasefile=test.txt
 
Hope you unstertand what I mean. Thank you

hneemann, 2019-09-30:
Sounds good to me. I'll implement that.

hneemann, 2019-09-30:
@d-teif 
Please try the latest [pre release v0.23-110](https://infdigital.dhbw-mosbach.de/) version.

To run the tests in the file `Circuit.dig`:
```
java -jar Digital.jar -t Circuit.dig
```

To check the circuit `Circuit.dig` with the tests given in the file `Tests.dig`:

```
java -jar Digital.jar -t Circuit.dig -c Tests.dig
```

hneemann, 2019-09-30:
@d-teif 
If you want to suppress the splash screen please use:

```
java -cp Digital.jar de.neemann.digital.gui.Main -t Circuit.dig
```

To check the circuit `Circuit.dig` with the tests given in the file `Tests.dig`:

```
java -cp Digital.jar de.neemann.digital.gui.Main -t Circuit.dig -c Tests.dig
```

closed 2019-10-01 by hneemann

hneemann, 2019-10-01:
@d-teif
Please try again the latest [pre release](https://infdigital.dhbw-mosbach.de/) version.

I decided to change the API a little bit:

```
java -cp Digital.jar de.neemann.digital.testing.CommandLineTester [dig file to test] [[optional dig file with test cases]]
```

d-teif, 2019-10-01:
Great ! Thanks, it work very well with multiple test case in same file.
1st version is more easy with -c and -t.

Regards 
D

hneemann, 2019-10-01:
> 1st version is more easy with -c and -t.

I agree! But the splash screen is flickering around all the time, so i decided not to start the jar via the manifest, but by the class name, which avoids the splash screen showing up.
And if the line in the bash script gets a bit longer, it doesn't matter much.

d-teif, 2019-10-01:
You're right it's just for script.

commit 841b190b (2019-10-01): adds a command line interface for testing circuits, closes #353