hneemann/Digital #355: Feature request: evaluate test cases from command line

jsegarraf, 2019-10-02:
I'd like to be able to evaluate a circuit from the command line using a specific test case file (not included in the circuit), and obtain the results in text. Sort of launching `java -jar Digital.jar -test <ComponentName> <testdata.txt> <circuit.dig>`. In this way a non-interactive script could be run to autoevaluate a component and generate feedback depending on the results.

Having many students, it is not feasible to manually add a test case, run it, and generate a feedback report for each component/circuit/student.

This feature is present in logisim-evolution with -testvector.

hneemann, 2019-10-02:
This feature was recently implemented. See #353

jsegarraf, 2019-10-02:
Thanks, I revised all issues some days ago, but I did not expect a so fast development. I'll try it.

closed 2019-10-02 by jsegarraf