hneemann/Digital #356: Faster pom.xml?

SmallRoomLabs, 2019-10-09:
Could you *please* make me a super-lean pom file that just compiles the code and makes the jar file. Noting else. No zips, no exes, no docs, no tests not even the checkstyle.

Currently I've disabled the tests with -Dmaven.test.skip=true and also prevented the maven files to be downloaded every time with -o.  But it still takes 4.5 minutes on my macbook (or about 6 minutes for the next compile since the CPU is throttled by the heat from the previous compile).

Being not 100% confident in Java and the design of Digital I tend to compile, and recompile and test quite a lot.  And the 5 minutes dev cycles get boring :zzz: real fast... ;-(

hneemann, 2019-10-09:
I'm not sure if this is a Java/Maven problem. :smile:

The command
```
mvn install -Dmaven.test.skip=true
```
requires 12s to finish on my machine.

The pom file attached reduces the build time to 2.5s.

[pom.xml.zip](https://github.com/hneemann/Digital/files/3708940/pom.xml.zip)

SmallRoomLabs, 2019-10-09:
Hmmm... So my machine is about 30 times slower than yours ;-)   I guess running it in a docker container with a mounted disk to the Mac is not a good idea.  I've never noticed any slowdowns like that when compiling golang code.

But the new pom did improve things a lot!  Now I get less than 2 minutes. And it only compiles 70-90 source files instead of the 700-something it always did before. 

```
[INFO] Scanning for projects...
[INFO] 
[INFO] ---------------------< de.neemann.digital:digital >---------------------
[INFO] Building Digital 1.0-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- git-commit-id-plugin:2.2.1:revision (get-the-git-infos) @ digital ---
[INFO] 
[INFO] --- maven-resources-plugin:2.6:resources (default-resources) @ digital ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Copying 129 resources
[INFO] 
[INFO] --- maven-compiler-plugin:3.1:compile (default-compile) @ digital ---
[INFO] Compiling 86 source files to /root/Digital/Digital/target/classes
[INFO] 
[INFO] --- maven-resources-plugin:2.6:testResources (default-testResources) @ digital ---
[INFO] Not copying test resources
[INFO] 
[INFO] --- maven-compiler-plugin:3.1:testCompile (default-testCompile) @ digital ---
[INFO] Not compiling test sources
[INFO] 
[INFO] --- maven-surefire-plugin:3.0.0-M1:test (default-test) @ digital ---
[INFO] Tests are skipped.
[INFO] 
[INFO] --- maven-jar-plugin:2.5:jar (default-jar) @ digital ---
[INFO] Building jar: /root/Digital/Digital/target/digital-1.0-SNAPSHOT.jar
[INFO] 
[INFO] --- maven-assembly-plugin:2.2-beta-5:single (make-assembly) @ digital ---
[INFO] META-INF/ already added, skipping
[INFO] META-INF/MANIFEST.MF already added, skipping
[INFO] META-INF/ already added, skipping
[INFO] META-INF/MANIFEST.MF already added, skipping
[INFO] org/ already added, skipping
[INFO] org/xmlpull/ already added, skipping
[INFO] org/xmlpull/v1/ already added, skipping
[INFO] org/xmlpull/v1/XmlPullParser.class already added, skipping
[INFO] org/xmlpull/v1/XmlPullParserException.class already added, skipping
[INFO] META-INF/ already added, skipping
[INFO] META-INF/MANIFEST.MF already added, skipping
[INFO] org/ already added, skipping
[INFO] META-INF/maven/ already added, skipping
[INFO] META-INF/MANIFEST.MF already added, skipping
[INFO] META-INF/ already added, skipping
[INFO] META-INF/maven/ already added, skipping
[INFO] org/ already added, skipping
[INFO] META-INF/ already added, skipping
[INFO] META-INF/MANIFEST.MF already added, skipping
[INFO] org/ already added, skipping
[INFO] org/slf4j/ already added, skipping
[INFO] META-INF/maven/ already added, skipping
[INFO] META-INF/maven/org.slf4j/ already added, skipping
[INFO] Building jar: /root/Digital/Digital/target/Digital.jar
[INFO] META-INF/ already added, skipping
[INFO] META-INF/MANIFEST.MF already added, skipping
[INFO] META-INF/ already added, skipping
[INFO] META-INF/MANIFEST.MF already added, skipping
[INFO] org/ already added, skipping
[INFO] org/xmlpull/ already added, skipping
[INFO] org/xmlpull/v1/ already added, skipping
[INFO] org/xmlpull/v1/XmlPullParser.class already added, skipping
[INFO] org/xmlpull/v1/XmlPullParserException.class already added, skipping
[INFO] META-INF/ already added, skipping
[INFO] META-INF/MANIFEST.MF already added, skipping
[INFO] org/ already added, skipping
[INFO] META-INF/maven/ already added, skipping
[INFO] META-INF/MANIFEST.MF already added, skipping
[INFO] META-INF/ already added, skipping
[INFO] META-INF/maven/ already added, skipping
[INFO] org/ already added, skipping
[INFO] META-INF/ already added, skipping
[INFO] META-INF/MANIFEST.MF already added, skipping
[INFO] org/ already added, skipping
[INFO] org/slf4j/ already added, skipping
[INFO] META-INF/maven/ already added, skipping
[INFO] META-INF/maven/org.slf4j/ already added, skipping
[WARNING] Assembly file: /root/Digital/Digital/target/Digital.jar is not a regular file (it may be a directory). It cannot be attached to the project build for installation or deployment.
[INFO] 
[INFO] --- maven-install-plugin:2.4:install (default-install) @ digital ---
[INFO] Installing /root/Digital/Digital/target/digital-1.0-SNAPSHOT.jar to /root/.m2/repository/de/neemann/digital/digital/1.0-SNAPSHOT/digital-1.0-SNAPSHOT.jar
[INFO] Installing /root/Digital/Digital/pom.xml to /root/.m2/repository/de/neemann/digital/digital/1.0-SNAPSHOT/digital-1.0-SNAPSHOT.pom
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  01:44 min
[INFO] Finished at: 2019-10-09T20:33:16Z
[INFO] ------------------------------------------------------------------------
```

closed 2019-10-10 by SmallRoomLabs

SmallRoomLabs, 2019-10-10:
Oh yes.... Running maven//javac in a Docker container was definitely a bad idea. Tonight I installed a native version of maven direct in OSX instead, and running your fast pom now only takes 6 seconds.  🏎  

That's quite a difference from the original 6 minutes I had to suffer through in the beginning. Well, I'll learned something new at least ;-)