hneemann/Digital #359: The measurements graph

SmallRoomLabs, 2019-10-15:
This looks a little bit strange to me, but I'm convinced that there is a perfectly find explanation for it. ^__^

At the graph it looks like the Clk stops for a while there's a (manually toggled) transition on a In-component or a Buttton. 

[image]

SmallRoomLabs, 2019-10-15:
Even two Clk sources with a ratio of a power-of-2 exhibits this strange phenomenon ...

[image]

hneemann, 2019-10-16:
The explanation is simple: In Digital there is no concept comparable with "time". There is only a modification of the model whose effects are calculated. This means that the x-axis does not show the time, but the consecutive number of the modification.

SmallRoomLabs, 2019-10-16:
Ah...  Maybe it's only the Clocks that makes everything screwed up somehow.   Replacing the two slower clocks with flipflops to divide the 4Hz down and everything looks reasonable again.

[image]

hneemann, 2019-10-16:
Think of it that way: Each time interval in which there was no signal change is displayed with the same length.  
And when different clocks are used, they will never be perfect in phase. However, if you use a synchronous counter, the output bits are perfectly in phase. And even if an asynchronous counter is used, the phase shift is only in the order of the gate delay, which is not shown in this graph.

SmallRoomLabs, 2019-10-16:
Yes, it is the Clocks not being in phase that was throwing me off a bit.  

I had assumed that the Clocks would update (as relevant for each of them) at each discrete step in the simulation.  And with discrete step I mean when the states propagated and fully settled down after a change somewhere. 

I'd love to see a more of a logic analyzer-like graph as well though. With the full state sampled at both edges of the (highest frequency) clock.  Then it would be like a regular time-domain LA or oscilloscope display.

I did (almost) this is my PDP8 with a counter and an eeprom that at each clock cycle stored data in the eeprom and incremented the counter so I could then later dump and analyze the collected data....

hneemann, 2019-10-17:
> I'd love to see a more of a logic analyzer-like graph as well though. With the full state sampled at both edges of the (highest frequency) clock. Then it would be like a regular time-domain LA or oscilloscope display.

But essentially that's exactly what happens.
Only that no data is stored if there was no change. But I agree with you:  Although this optimization is often useful, it can also be confusing, because it leads to the fact that you can no longer see from the diagram where more or less time has passed.

SmallRoomLabs, 2019-10-17:
Right...  I'm closing this for now, because I now know what is happening and hopefully I can dechiper the data in a better way with this knowledge.  

And I've also given you a hint that it is kinda confusing even for an experienced guy like me (or maybe it is confusing just because I'm old and experienced and already have assumptions of what the graph should represent and look like). ;-)

closed 2019-10-17 by SmallRoomLabs