hneemann/Digital #361: Internal race condition issue

SmallRoomLabs, 2019-10-17:
This is a strange one.  In my design I have an edge detector and a DFF used as a clock domains synchronizer (I know it should preferably be two or three DFFs to be fully protected in a real life situation, but this will suffice for now).   In front of the DFF I have a S/R to capture the short pulse from the edge detector.

In this test I have the synchronizer connected to the direct base clock instead of a /16 clock as my real circuit (an UART). This is simply to be able to trigger the bug faster without a lot of clicking on the STB button.

The clock that now drives both the edge detector and the DFF can in this test be switched between direct to the Osc-component or the osc simply divided by two.

As I see it the circuit should not behave differently if I run it on 500Hz or 250 Khz since all parts of it is clocked by the same signal.

But If I select the direct Osc clock I can trigger an oscillation-error within a few clicks on on STB button.  Switching over to the div-by-2 clock I can click until the cows come home without any problems.

It's like this reardless if I set the Osc to "maximum" or something low like 5KHz.

I guess that this is an artifact caused by the simulated gate delays being calculated from the Osc-clock. Which is reasonable.  But in this case it causes the simulation to behave differently if I just patch in a div-by-2 between the oscillator and the rest of the design.  In real life the race/oscillation condition would be present regardless.

Attaching the file here: [RaceIssue.dig.txt](https://github.com/hneemann/Digital/files/3741124/RaceIssue.dig.txt) (remove the .txt extension)

[image]

hneemann, 2019-10-18:
That's not a rece condition in the java code. It's a race condition in the circuit.
To understand what is happening here, you have to know that in Digital all components are equally fast, i.e. have the same propagation time. In particular, a D-FF is as fast as a NAND gate.
This means that the pulse from the edge detector only has the duration of one propagation time when STB and CLK simultaneously change to high..
But the RS-FF needs two propagation times to switch, because two NAND gates have to react.
The switching pulse already disappears while the RS-FF has switched halfway. In this case the RS-FF gets stuck in an oscillation.
If you slow down the D-FF in the edge detector by inserting a delay element at the output, this can't happen anymore.

hneemann, 2019-10-18:
> What I clumsily tried to convey is that the simulation ideally should't work differently whether the entire circuit is clocked by a Clk-component directly or by Clk/2. A real circuit with a race would in most cases have the race regardless if it is clocked at 10MHz or 5MHz.

That's right. But this is an asynchronous event. A keystroke that takes place at any time. If the frequency is very high, there are many edges and it is easier to hit one. The slower the frequency, the harder it will be to create such a problem. Sometimes so hard that it seems as if the circuit is faultless.

But in the end, it remains a simulation. In many points it deviates from a real circuit. For me, who knows the simulation and the limits very well, it is rather surprising how close the simulation comes to some real problems. And for other problems, the simulation is - as expected - not verry close to reality.

closed 2019-10-18 by SmallRoomLabs