hneemann/Digital #364: Runtime error in Undo

SmallRoomLabs, 2019-10-18:
It didn't crash the program, but undo stopped working (or became very unresponsive)  after this so I restarted it. I didn't have an enough large line-buffer in my terminal window to capture more than this:

```
        at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547)
        at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
        at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2762)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4840)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:389)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
        at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
        at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:389)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
        at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
de.neemann.digital.draw.elements.PinException: No output connected to a wire ([MQ]/[in; Tunnel, 0-11; Splitter, 0-11; Splitter, in; Tunnel]). The state of the wire is undefined.
        at de.neemann.digital.draw.model.Net.interconnect(Net.java:163)
        at de.neemann.digital.draw.model.ModelCreator.createModel(ModelCreator.java:277)
        at de.neemann.digital.gui.Main.createAndStartModel(Main.java:1346)
        at de.neemann.digital.gui.Main.access$4300(Main.java:94)
        at de.neemann.digital.gui.Main$RunModelState.enter(Main.java:1319)
        at de.neemann.digital.gui.Main$RunModelState.enter(Main.java:1310)
        at de.neemann.digital.gui.state.State$1.actionPerformed(State.java:82)
        at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
        at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
        at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
        at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
        at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279)
        at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:297)
        at java.desktop/java.awt.Component.processMouseEvent(Component.java:6632)
        at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
        at java.desktop/java.awt.Component.processEvent(Component.java:6397)
        at java.desktop/java.awt.Container.processEvent(Container.java:2263)
        at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5008)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4840)
        at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
        at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547)
        at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
        at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2762)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4840)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:389)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
        at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
        at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:389)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
        at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
java.lang.RuntimeException: internal error in undo
[... 796 lines cut ...]
 Attributes of component 'Tunnel' modified.
 Selection moved.
 Component 'Tunnel' moved or rotated.
 Selection moved.
 Inserted wire.
>
        at de.neemann.digital.undo.UndoManager.createTrace(UndoManager.java:133)
        at de.neemann.digital.undo.UndoManager.undo(UndoManager.java:117)
        at de.neemann.digital.gui.components.CircuitComponent.undo(CircuitComponent.java:494)
        ... 38 more
Caused by: de.neemann.digital.undo.ModifyException: internal error: Element not found!
        at de.neemann.digital.gui.components.modification.ModificationOfVisualElement.getVisualElement(ModificationOfVisualElement.java:74)
        at de.neemann.digital.gui.components.modification.ModifyMoveAndRotElement.modify(ModifyMoveAndRotElement.java:36)
        at de.neemann.digital.gui.components.modification.ModifyMoveAndRotElement.modify(ModifyMoveAndRotElement.java:17)
        at de.neemann.digital.undo.UndoManager.undo(UndoManager.java:112)
        ... 39 more

(java -jar tools/Digital-SRL.jar sim/PDP8-X.dig &)
Matss-MacBook-Pro-6:PDP8-X mats$ WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.thoughtworks.xstream.core.util.SerializationMembers (file:/Users/mats/Documents/Projects/PDP8-X/tools/Digital-SRL.jar) to method java.awt.geom.AffineTransform.readObject(java.io.ObjectInputStream)
WARNING: Please consider reporting this to the maintainers of com.thoughtworks.xstream.core.util.SerializationMembers
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

```

hneemann, 2019-10-19:
In your home folder are files named `Digital_2019-10-12_12-57-18.log` or similar.
Could you please zip them and attach them to this issue?

SmallRoomLabs, 2019-10-19:
[logs-Digital_2019-10-19.zip](https://github.com/hneemann/Digital/files/3746376/logs-Digital_2019-10-19.zip)

hneemann, 2019-10-19:
I'm guessing you don't use generic circuits?

SmallRoomLabs, 2019-10-19:
If by Generic Circuit you mean Embedded/Subcircuits with the "Circuit is Generic" checked then the answer is no.  I have not used that feature at all. Plenty of regular components with the width set to (usually) 12 though.

I originally had everything on one schematic, but now when it's debugged and working (more or less properly) I've started to break out parts into subcircuits in a size that can be fit on a single pcb. And I also change starting to change _(ha!  that is what you mean!)_ the generic parts into real 74-series parts (that still can be purchased - i stay away from the unobtaniums).

SmallRoomLabs, 2019-10-19:
The green boxes are the subcircuits so far - they have in total about 90 ICs.

[image]

hneemann, 2019-10-19:
Do you remember doing anything unusual? The undo error occurred because a component changed its position without creating an entry in the undo system. Therefore, this change could not be restored by the undo system, which was the cause of the error.
The problem is that I have no idea how a change could be made by bypassing the undo system.