hneemann/Digital #366: Feature request: start tutorial from menu

jsegarraf, 2019-10-25:
As far as I know, the tutorial cannot be started again if it has been completed. For students working on the same computer at different times, this means that only the first one will see the tutorial.

It would be perfect if the tutorial could be started again from the gui, for instance from the View menu.

Currently, the only option I see to restart the tutorial is to remove the configuration file .digital.cfg in Linux (I don't know the file names and locations in other systems), but it is not a good approach.

hneemann, 2019-10-25:
Please try the latest [pre release build](https://infdigital.dhbw-mosbach.de/).

closed 2019-10-25 by hneemann

jsegarraf, 2019-10-25:
Works perfectly.
Thank you very much

commit a2373e39 (2019-10-25): Adds a "Start Tutorial" entry to the view menu, closes #366