hneemann/Digital #369: Can the Text element display none English language?

itviewer, 2019-10-30:
I want to add some comments to the circuit, but the Text element does not support displaying Chinese.

hneemann, 2019-10-30:
Yes it can. But Java supports by default only code points below U+FFFF. 
Here are some CJK Unified Ideographs above U+4E00 on my machine:

[image]

itviewer, 2019-10-31:
Does the application will  set the appropriate font automatically? I tried  some methods, but my font still shows as a box.
[image]
Then I tried to hard code the font setting  in file src/main/java/de/neemann/digital/draw/graphics/Style.java line 170 from
`        font = new Font("Arial", fontStyle, fontSize);
`
to
`        font = new Font("SimSun", fontStyle, fontSize);
`
"SimSun"  is the font I need for Chinese language. After that, the font can be displayed normally.

[image]

Can it be achieved without hard coding?

hneemann, 2019-10-31:
I have not changed any code at all to create the screenshot above.
But maybe it is OS or JRE/JDK dependent. I have used Debian with OpenJDK 11 installed.
To me it seems like a Java/JRE/JDK configuration problem, and not a bug in Digitals source code.
There are several web sites out there that deal with this problem. 
But i can not figure out myself what the problem is, because on my system it works out of the box without any problems.
Could you please attach the circuit above containing the Chinese text? That would enable me to do a few more experiments.

hneemann, 2019-10-31:
Also a more complex text is working on my machine out of the box (created with the google translator):
[image]

itviewer, 2019-10-31:
I think the problem should be Java's own support for CJK fonts. I use Ubuntu with openjdk-8.  
My Google solution is to put the font in the `jre/lib/fonts/fallback` directory, but it doesn't work 
 for me.
Maybe JDK 11 should be friendly to CJK fonts. I will to try it later, thanks.

closed 2019-10-31 by itviewer

hneemann, 2019-10-31:
Today I played around a little on a Windows computer. The CJK fonts didn't work there either. 
I coulf fix the problem by replacing the line you mentioned above from
```
font = new Font("Arial", fontStyle, fontSize);
```
to
```
font = new Font(null, fontStyle, fontSize);
```
Could you please try if that works for you?

commit 872c472b (2019-10-31): Fixes a problem with Chinese text rendering, closes #369