hneemann/Digital #375: HDL export and async RS FF's

SmallRoomLabs, 2019-11-06:
After a break in designing the PDP8 I wanted to see roughly how much of a 1K ICE40 my different modules would occupy.   After installing the required toolchains and playing around a bit I ran into #161, but that was easy to fix in each module.

Worse is that asynchronous RS flipflops aren't supported in Digital, and I use them both "loose" in my design and they are also used in the DIP versions of some ICs like the 74373.

Is there some particular reason they're not supported?  From my limited knowledge and recent study of verilog it seems rather straightforward to implement async functionality in modules.  (Of course race conditions and metastabillties comes into play, but that's a fact of life that one has to live with... ^__^ )

hneemann, 2019-11-07:
It is not easy to support combinatorial loops because the implementation depends on the synthesis tool used. In the Xilinx tools, for example, one of the nets in the loop must be specially marked in the circuit constraints, otherwise the synthesis terminates with an error.

SmallRoomLabs, 2019-11-07:
Oh, I though that verilog was (at least at the basic level) standardized enough between the manufacturers that only things like hard blocks would differ between them.

So this means that it's not totally easy to move simple designs between manufacturers. That sucks a bit.

hneemann, 2019-11-07:
Some people would not agree with the statement that a combinatorial loop in an FPGA is one of the basic things.  :-)

SmallRoomLabs, 2019-11-07:
Ok, I see.  So anything asynchronous like latches or async resets on FFs are to be avoided as far as possible in an fpga - this in contrast with a discrete "physical" implementation where those elements are quire common and accepted (with a bit of care and due dilligence)

I love these discussions where I learn new things related to my projects - it's almost so that I fondly look back to school where this was a common occurence :-)

nikikust, 2019-11-07:
sorry for my question here, but why is HDL for Division not supported in digital? Thanks and sorry again :)

hneemann, 2019-11-07:
@nikikust 
Division's a pretty complex operation. If you simply use the division operator in an HDL, the code is usually not synthesizable.
There are a number of [algorithms](https://en.wikipedia.org/wiki/Division_algorithm) that can be used instead. 
However, it is important to consider whether you want to choose a fast method that consumes a lot of resources on the FPGA or a slow, resource-saving method.
There is no one-fits-all solution.

nikikust, 2019-11-08:
Should i made these algorithms in digital by supported elements? or can i add some Verylog or VHDL code to Div element?

hneemann, 2019-11-08:
@nikikust 
Both is possible: If you use the "Extern" component in the misc menu, you can add Verilog or VHDL code to the simulation. When the circuit is exported to Verilog or VHDL, this code is used in the exported output. However, if you want to simulate such a circuit, you must install IVerilog or GHDL to simulate the circuit's HDL code, and the interface to these programs makes the simulation quite slow.

If you construct the divider with the built-in components, the simulation is still fast, and you can export to both Verilog and VHDL. But in this case you have to build the divider yourself, because there is no divider example out there while hdl implementations are available.

closed 2019-11-09 by hneemann