hneemann/Digital #378: MUX missing selector

equokka, 2019-11-12:
[image]
I'm fairly new to this so I'm sure I'm misunderstanding something, either about how circuitry is drawn or how this software does this, but shouldn't a 4:1 multiplexer have 2 selector inputs? 

Here's an example of what I mean, from Wikipedia:
[image]

hneemann, 2019-11-12:
You're right about that. But Digital uses the concept of data buses. So that a circuit does not become too confusing, several bits can be combined to a single wire.
Imagine what a 32-bit CPU would look like if there were always 32 individual wires to draw.
The select input is always represented as a single input dot, but the wire connected there represents two bits.
See also #371

equokka, 2019-11-12:
May I ask how the splitter (merger in this case?) works?
[image]
The above setting gives me an error saying the multiplexer needs 2 bits but only 1 bit was found. If I change the "Output splitting" to 3 or 4, it instead gives me an error message saying 3, 4 bits were found...
How do I have this component merge two signals into a single 2bit wire?

hneemann, 2019-11-12:
I think the splitter is configured correctly. I think the multiplexer is the problem. The multiplexer is not only switched with a bus, it can also use buses as data inputs and outputs. In the screenshot at the top, the multiplexer seems to have been configured for a 2-bit data bus. However, it is wired with 1-bit signals.

closed 2019-11-14 by hneemann