hneemann/Digital #380: Please make it possible that the dialog to rename all tunnels of the same name can be switched off!

rdu, 2019-11-14:
I like the feature to rename all tunnels of the same name, when I rename a single tunnel. Sometimes however, it bothers me a lot. Especialy if I have to rename a lot of tunnels without having the intention to rename all the others with the same name. Please give me/us the option to turn the dialog off. Maybe with a modifier key?

hneemann, 2019-11-14:
The escape key closes the dialog (at least on Debian/Gnome).

rdu, 2019-11-14:
I know I can do that, but I have renamed 20 or more tunnels by mistake several times without wanting to. I would be really happy if it could be deactivated. In an old version the function wasn't there yet - I could work more effectively there. But the function itself is really cool - I would just like to deactivate it somehow (temporarily).

hneemann, 2019-11-15:
Could you please explain your use case in a bit more detail?
How did it happen that you have to convert one net into several different nets?
Maybe there is a better solution than switching off the automatic renaming.

SmallRoomLabs, 2019-11-15:
Before I discovered that I can rotate a tunnel with R during placement I used to frequently copy and rename tunnels that was nearby (and in the right orientation) instead of navigating the menus, selecting, placing and then separately rotating a new tunnel.

It actually still happens that I copy/paste/rename tunnels since that feels faster than doing the menu. And of course it happens that I every now and then click on the wrong button in the "change all" -dialog.

So I've been wishing for a "change all" -turnoff as well.  But a better solution would probably be to be able to have shortcuts for common items.  For me the tunnels, constant 0/1 would be sufficient. I guess that other have other preferences for shortcuts.  

But on the other hand the bar of "last x used components" is probably good enough for this purpose anyways, but I tend to forget about it and go menuing most of the time :-(

hneemann, 2019-11-15:
@SmallRoomLabs 
That's interesting! It's easy to add a shortcut [T] to insert a tunnel.
But if you don't like the menu because it's slow, what about the components tree view which can be activated by pressing [F5]?

rdu, 2019-11-15:
I copy and paste multiple tunnels and rename it afterwards. This workflow ist much faster for me than place a new tunnel every time and rename it. Especially when I only have to rename some parts of the label.

Nevertheless this ist just a tiny feature request for the comfort. Needless to say all in all it's not really a big deal. I like this software very much!

SmallRoomLabs, 2019-11-16:
@hneemann , I'm usually very conservative when it comes to screen real estate.  I've used a 13" macbook as my main machine for the last 5 years so I really have to keep just the waste to a minimum.

So I disregarded the F5 tree right away, but when I looked at it now again I see it's not too bad. So I'll give it a try when designing the next time.  Being able to quickly hide/unhide it with F5 helps a lot.

But a T shortcut for tunnels would be nice.  I'm not sure whether being automatically dropped into the name-dialog after you placed a tunnel would be a good idea or not. That might be just annoying or a good timesaver - who knows?

And if you're adding the T shortcut, then you might consider having a shortcut for toggling the Constant between 0 and 1 during placement. Maybe T can be reused, or space or maybe the +/-...

hneemann, 2019-11-16:
The latest [pre release](https://infdigital.dhbw-mosbach.de/) contains the new shortcuts.

hneemann, 2019-11-16:
How about a "Rename" dialog:
You select a bunch of components and via the menu you can open a single dialog that displays all the labels used in the selected components and allows you to edit them.

closed 2019-11-26 by hneemann