hneemann/Digital #382: Changing numeration of cells in arnaugh Maps 3 vars and 4 var

helderdaniel, 2019-11-18:
Hi,

helderdaniel, 2019-11-18:
Hi,

I used Logisim for some time and found Digital sometime ago.
Great work on Digital and it is getting better every day :)

However, I miss the numbering of kmaps cells as Logisim does:

[image]

[image]

I am used to numbering kmaps cells like this.
I found that Digital uses another order (tried with last version: Nov 16, 2019, ce4ed8f):

[image]

[image]

So I cloned the repo, changed the code in the file: KarnaughMap.java to number as I like (complete changed file attached, had to change the extension to *.txt to upload).

I just changed the order of the inverted variables as shown below and compiled with: `mvn install`

```
//(...)
case 3:
                for (int row = 0; row < 2; row++)
                    for (int col = 0; col < 4; col++)
                        cells.add(new Cell(row, col));
                addToRow(0, 4, 0, true);
                addToRow(1, 4, 0, false);
                addToCol(0, 2, 1, true);
                addToCol(1, 2, 1, true);
                addToCol(2, 2, 1, false);
                addToCol(3, 2, 1, false);
                /*
                addToCol(0, 2, 2, false);
                addToCol(1, 2, 2, true);
                addToCol(2, 2, 2, true);
                addToCol(3, 2, 2, false);
                */
                addToCol(0, 2, 2, true);
                addToCol(1, 2, 2, false);
                addToCol(2, 2, 2, false);
                addToCol(3, 2, 2, true);

                headerLeft = new Header(0, true, false);
                headerTop = new Header(1, true, true, false, false);
                //headerBottom = new Header(2, false, true, true, false);
                headerBottom = new Header(2, true, false, false, true);
                headerRight = null;
                break;
            case 4:
                for (int row = 0; row < 4; row++)
                    for (int col = 0; col < 4; col++)
                        cells.add(new Cell(row, col));

                addToRow(0, 4, 0, true);
                addToRow(1, 4, 0, true);
                addToRow(2, 4, 0, false);
                addToRow(3, 4, 0, false);
                /*
                addToRow(0, 4, 1, false);
                addToRow(1, 4, 1, true);
                addToRow(2, 4, 1, true);
                addToRow(3, 4, 1, false);
                */
                addToRow(0, 4, 1, true);
                addToRow(1, 4, 1, false);
                addToRow(2, 4, 1, false);
                addToRow(3, 4, 1, true);

                addToCol(0, 4, 2, true);
                addToCol(1, 4, 2, true);
                addToCol(2, 4, 2, false);
                addToCol(3, 4, 2, false);
                /*
                addToCol(0, 4, 3, false);
                addToCol(1, 4, 3, true);
                addToCol(2, 4, 3, true);
                addToCol(3, 4, 3, false);
                */
                addToCol(0, 4, 3, true);
                addToCol(1, 4, 3, false);
                addToCol(2, 4, 3, false);
                addToCol(3, 4, 3, true);

                headerLeft = new Header(0, true, true, false, false);
                //headerRight = new Header(1, false, true, true, false);
                headerRight = new Header(1, true, false, false, true);
                headerTop = new Header(2, true, true, false, false);
                //headerBottom = new Header(3, false, true, true, false);
                headerBottom = new Header(3, true, false, false, true);
                break;
//(...)
```

However I had to disable one of the tests in the file: KarnaughMapTest.java (full changed file attached),

`public void testBUG_1()`

so that it passes all tests. (Changing just the code to reorder vars in 3 vars kmaps does not trigger this testBUG_1 error).

```
//(...)
   // in 4x4 map a 8 cell block is drawn in wrong orientation
    public void testBUG_1() throws IOException, ParseException, KarnaughException {
/*    Expression exp = new Variable("D");
        KarnaughMap c = new KarnaughMap(Variable.vars(4), exp);

        assertEquals(1, c.size());

        KarnaughMap.Cover co = c.iterator().next();
        assertTrue(co.isDisconnected());
        assertFalse(co.onlyEdges());
        assertTrue(co.isVerticalDivided());

        exp = new Variable("B");
        c = new KarnaughMap(Variable.vars(4), exp);

        assertEquals(1, c.size());

        co = c.iterator().next();
        assertTrue(co.isDisconnected());
        assertFalse(co.onlyEdges());
        assertFalse(co.isVerticalDivided());
*/

//(...)
```

I am running the 2 versions, and as far as I test them the changed version seems to work as the original.
It creates the 3 and 4 vars kmaps with the reordered variables and simplifies the expression as expected.

However, I am concerned with that disabled unit test.
I did not get what the comment on that test means:

>    // in 4x4 map a 8 cell block is drawn in wrong orientation

Can you tell me if the original order of variables in Digital kmaps is really needed for proper operation?

thanks,

Helder

[KarnaughMap.java.txt](https://github.com/hneemann/Digital/files/3857649/KarnaughMap.java.txt)
[KarnaughMapTest.java.txt](https://github.com/hneemann/Digital/files/3857660/KarnaughMapTest.java.txt)

hneemann, 2019-11-18:
There are more than 1000 different ways to draw a 4x4 K-map.
Why do you care which one is used? They all lead to the same result.

hneemann, 2019-11-18:
The k-map is not used to calculate the minimized equation. The minimization is done with the method of Quine & McCluskey.
The k-map is only drawn for illustrative purposes using the already minimized equation.
Even if the k-map is drawn completely wrong, the result of the minimization is not affected.

helderdaniel, 2019-11-18:
> There are more than 1000 different ways to draw a 4x4 K-map.
> Why do you care which one is used? They all lead to the same result.

Yeah,
I was thinking in counters.
For a gray counter numbering like this seems easier to follow.
For a binary counter it is not that easy but it still seems easier. Of course that is for me ...
For other projects another numbering might be best or it does not even matter.

[image]

```

The k-map is not used to calculate the minimized equation. The minimization is done with the method of Quine & McCluskey.
The k-map is only drawn for illustrative purposes using the already minimized equation.
Even if the k-map is drawn completely wrong, the result of the minimization is not affected.
```

ok,

thanks

jsegarraf, 2020-03-11:
Hi,

I would like this issue to be reopened as a feature request, in order to change the ordering of the cells as @helderdaniel proposes.

Although any ordering may be valid, as far as I know most books use the proposed ordering. Thus, from a student perspective, it would be helpful seeing a representation that visually matches what they expect.

Thanks

hneemann, 2020-03-11:
@jsegarraf 
Maybe you are right!

helderdaniel, 2020-03-11:
Hi, 

At the time I did a fork and implemented the changes proposed in the original post. 
As far as I can tell it seems to be working fine since then. 

However as also mentioned in the post I had to disable one of the tests from file KarnaughMapTest.java:

public void testBUG_1()

so that it passes all tests. 

I also found that changing just the code to reorder vars in 3 vars kmaps does not trigger this testBUG_1() unit test. It is a issue just when changing the code for 4 var maps. 

That was my main concern at the time, since I didn't figure out what this unit test is for.

hneemann, 2020-03-11:
@helderdaniel 
As far as I  can see, your fork contains the original cell order. 

The test case is not a problem: If a variable is inverted in the k-map, you have to invert the variable also in the test cases, which depend on the ordiring in the k-map.

hneemann, 2020-03-11:
@helderdaniel, @jsegarraf 
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/).

helderdaniel, 2020-03-11:
74804 (NAND), 74805 (NOR), 74808 (AND),…

On Wed, 11 Mar 2020 at 10:36, Helmut Neemann <notifications@github.com>
wrote:

> @helderdaniel <https://github.com/helderdaniel>, @jsegarraf
> <https://github.com/jsegarraf>
> Please try the latest pre release <https://infdigital.dhbw-mosbach.de/>.
>
Looks ok, thanks

As far as I can see, your fork contains the original cell order.

Right!
I didn't push the proposed code in my original post since the issue was
closed.

What I pushed was just the addition of the hex 2-input gate chips.
Sorry, it was some time ago ...

hneemann, 2020-03-11:
> Sorry, it was some time ago ...

Never mind! :smiley:

jsegarraf, 2020-03-12:
Pre-release works perfectly, thanks

commit c2c8eafa (2020-03-13): updates the release notes, closes #382
closed 2020-03-13 by hneemann