hneemann/Digital #384: Warning with java 11

d-teif, 2019-11-20:
Hello

Using java 11, i have the following message when i start Digital. Have you heard about this ?

Regard 

Didier 

java -jar Digital.jar 
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.thoughtworks.xstream.core.util.Fields (file:/home/AD/dteifret/DossierPartage/ADO/PRJ/Digital.jar) to field java.util.TreeMap.comparator
WARNING: Please consider reporting this to the maintainers of com.thoughtworks.xstream.core.util.Fields
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

hneemann, 2019-11-20:
I have already fixed this warning. 
Please try one of the latest [pre release](https://infdigital.dhbw-mosbach.de/) builds.

SmallRoomLabs, 2019-11-27:
I still get this in the latest build...

```
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.thoughtworks.xstream.core.util.SerializationMembers (file:/Users/mats/Documents/Projects/PDP8-X/Digital.jar) to method java.awt.geom.AffineTransform.readObject(java.io.ObjectInputStream)
WARNING: Please consider reporting this to the maintainers of com.thoughtworks.xstream.core.util.SerializationMembers
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
```
```
java --version
java 12.0.2 2019-07-16
Java(TM) SE Runtime Environment (build 12.0.2+10)
Java HotSpot(TM) 64-Bit Server VM (build 12.0.2+10, mixed mode, sharing)
Matss-MacBook-Pro-6:PDP8-X mats$ 
```

hneemann, 2019-11-27:
Does the warning appear immediately after starting?

SmallRoomLabs, 2019-11-27:
Yes, it's printed  a few seconds after doing `java -jar Digital.jar'

SmallRoomLabs, 2019-11-27:
Hmmm... That was a "truth with modifications".  If I let it open my design (which it does by default) the warning message is printed.  If I open a newly saved completely empty .dig -file I don't get the message.

hneemann, 2019-11-27:
I have already found the problem: It is caused by the [CTRL]+[n] stored views.

SmallRoomLabs, 2019-11-27:
Oh noes!  My favorite feature is making a mess!  ;-)

closed 2019-11-27 by hneemann

hneemann, 2019-11-27:
This one was tricky to fix.

Normally I try not to break old circuits in new versions of Digital.
Since the "store view" feature hasn't been released yet, this time it's different: It is no longer possible to open a circuit in which views are stored. These have to be removed from the XML file using a text editor. Sorry for that.

You can easily find the views: They are located at the top of the file and look like this: 

``` XML
    <entry>
      <string>view1</string>
      <transform serialization="custom">
        <transform>
          <default>
            <m00>0.863556338028169</m00>
            <m01>0.0</m01>
            <m02>-236.75</m02>
            <m10>0.0</m10>
            <m11>0.863556338028169</m11>
            <m12>55.4375</m12>
          </default>
        </transform>
      </transform>
    </entry>
```

See also #335

commit 8d1defdf (2019-11-27): fixed a reflective access issue caused by stored views, closes #384