hneemann/Digital #385: Hex file format

d-teif, 2019-11-20:
I'm  using mars for me CA course 
https://courses.missouristate.edu/KenVollmar/MARS/

and  create intel hex file with this tool. (join file).  Really an intel hex f
[Inst.hex.gz](https://github.com/hneemann/Digital/files/3868404/Inst.hex.gz)

ormat
I have message from Digital
[AWT-EventQueue-0] INFO de.neemann.digital.core.memory.importer.Importer - /home/AD/dteifret/DossierPartage/ADO/PRJ/Inst.hex: could not read native hex, try intel hex

Regard 

Didier

hneemann, 2019-11-20:
This is just an info message stating that the file is not the old logisim format, and therefore attempts to load the file as an Intel-HEX file.
And on my computer the file will be loaded as an Intel-HEX file without any errors.

d-teif, 2019-11-20:
thanks for this information

closed 2019-11-20 by hneemann