hneemann/Digital #388: Organize pins on custom parts

jdgabbard, 2019-11-29:
So, I just discovered the software today.  Seems real nice, although I haven't tested any of the VHDL or JEDEC exporting.  You're doing well!

I must ask that you add the ability to intuitively reorder the locations of pins on a custom part.  It seems the software randomly throws the inputs to a part at random places.

So to clarify, I designed several smaller schematics, saved them, and added them to the Lib folder (in Windows) inside of a new folder.  I am referring to these saved schematics as "parts" for the purpose of explaining this.  Digital recognizes that there are new parts in the drop down menu.  However, when I go to add these parts to a larger schematic, it seems as though it randomly picks a location to stick the inputs/outputs.

This is a headache when you have 3-4 signals that are related in some way, but are placed on opposite ends of the part.  It would be nice to be able to assign their location on the part without much more than a few clicks.

hneemann, 2019-11-29:
The menu "Edit->Order Inputs" and "Edit->Order Outputs" allows you to order the pins.
Alternatively, you can choose the shape "Layout" to display the embedded circuit (Edit->Circuit specific settings->Advanced->Shape). Then the pins are placed in the same order as the inputs and outputs in the circuit. In this case also pins at the top and bottom of the shape are possible.

jdgabbard, 2019-11-29:
That does the trick.  Thank you for the speedy response.  I was looking for the option in the larger schematic.  You apparently need to be in the schematic for the part.

closed 2019-11-29 by jdgabbard