hneemann/Digital #390: Signal Pass Through -> Export to Devices

Justinius, 2019-12-04:
I just found out about your project, but love that it allows easy export to GAL and other devices.

However, I was designing a simple circuit that took an input, used it as part of the calculation and then routed the signal to an output.

So Sig_In -> Sig_Out.

This shows up in the list of logic equations after an analysis, but when I looked at the generated PLD this was not in the file.

Justinius, 2019-12-04:
Related:  As I was testing the circuit I also noticed that the Digital analysis simplified the circuit to a pass through.  (I don't know how I missed that simplification for my low order bit.) But anyway that also shows up on the analysis as Out_0 = In_0  but again not exported in the PLD.

I know that since its an input I can just route the signal around the board, but I was hoping for automated flow through so everything would just be point to point along the circuit chain.

hneemann, 2019-12-04:
I looked at the code, and you're right: 
A simple alias "A=B" is intentionally removed and not used when creating the JEDEC file.
This feature was added by me as a feature in commit c0cfc0d941b763ecaac93248127d7275e15a96e9 three years ago before the first release, and to be honest: I can't remember why I implemented this function.
It's odd, of course, in a GAL where resources are so limited to connect one pin to another. But if the user wants that, why shouldn't it be possible?

Justinius, 2019-12-04:
Maybe it can be a configurable part of the export, or if detected prompts for input at that point.

hneemann, 2019-12-05:
I found out the reason for that feature. It's required to create a moore machine with missing output logic (Λ=S). This function ensures that in this case only one cell is allocated. Without that function a additional not needed pass through cell is allocated too)
I have added a new implementation that recognizes a simple pass through which is created by the user and in that case the removal is avoided.

Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) and let me know how it works. At the moment i have no JEDEC->programmer->GAL workflow at hand to try it myself.

Justinius, 2019-12-05:
Unfortunately do to traveling for work and holiday. I don't know when I'll be able to try it but I'll do that and comment as soon as possible.

hneemann, 2019-12-12:
closed in e923a9a440a3893cebff0088280db7ee22a757ef

Today I could test the new code with a GAL16v8D and it works fine.

closed 2019-12-12 by hneemann