hneemann/Digital #393: Drag connected wires when moving components

obiwanjacobi, 2019-12-10:
As one is creating a schematic moving components around is a regular task. Unfortunately I cannot find a way to keep the connected wires attached to the components when they are moved.

Logisim has nice support for this - something similar would be most useful.

hneemann, 2019-12-10:
If a component is selected by a click, only the component is moved without the wires. If instead a component is  selected by the selection rectangle, the component is moved including the wires.
However, no rerouting algorithm is implemented when a component is moved, including the wires.

obiwanjacobi, 2019-12-11:
Ah, ok. Practically though, it is not very useful. After moving you still are going to delete and redo these off-grid wires... 
This really prevents me from doing anything big in this app...

hneemann, 2019-12-11:
In my experience, the result of the autorouting in Logisim must also be manually reworked after a relocation of components in order to get a really good result.
So it's more a question of habit.

closed 2019-12-15 by hneemann

SaphireLattice, 2021-08-30:
While major component move does tend to result in a mess, it is much more useful for minor moves a few units around.