hneemann/Digital #394: Help with Bidirectional Inputs/Outputs

ProxyPlayerHD, 2019-12-13:
A bit of background info:

I made a CPU with Logisim and converted it to Digital so i can use it on an FPGA. for testing purposes i used the build-in RAM and ROM Components. but now i would like to test around with external Memory, mostly because using Memory on the FPGA itself uses up a lot of Block RAM and that just seems like a waste to me.

But RL RAM uses a bidirectional databus, so i need to interface it with a bidirectional pin in the simulator as well. Problem is, I'm not really sure how to go about doing that. my main idea is to split the bidirectional signal into 2 unidirectional signals (and a select line).

I came up with this design but as said i'm not sure if it would work as I'm not sure if the Input component can just behaive like a bidirectional pin without issues.
which is why i'm trying to ask here for a bit of help. (though i do know it works with the build-in bidirectional RAM component)
[image]

hneemann, 2019-12-13:
Up to now there is no easy way to implement that. 
It is not possible to create an INOUT port in a HDL export.
The only way is to add the required hdl code to the generated hdl file manually.

> but now i would like to test around with external Memory, mostly because using Memory on the FPGA itself uses up a lot of Block RAM and that just seems like a waste to me.

What do you want to save the block RAM for?

ProxyPlayerHD, 2019-12-13:
>Up to now there is no easy way to implement that.
aww dammit. not even as something similar to what i designed, except as a single component?
so you got something that can be named like a Input/Output and has 3 connections. 1 Data input, 1 Data output, and one selection input. if the selection input is 0 the components functions exactly like an Input, and when it's 1 it functions like an Output.

anyways i'll have to think of something, adding something to automatically generated code is not really easy.

>What do you want to save the block RAM for?

for one i just want to use less in case i port this project onto an FPGA with much less BRAM available than the one i'm currently using. and two, my future plans with this Computer include an VGA Interface and for that i want an internal Frame Buffer, which would already take up around 21600 bits of BRAM. also i'm not sure but i think ROM or LUT components also take up BRAM, which is why i have to somewhat rework my control unit, because a 4194304 bit large ROM/LUT for instruction decoding is a bit taxing on the FPGA

hneemann, 2019-12-13:
> but using more but smaller ROMs definetly decreased the time it takes to Analyse and 
> Synthesise the circuit (from 1.5h to 1.5min) so i still think that was worth it.

That's interesting! Which tool chain you are using?
But when you think about it, it's not surprising: A ROM is essentially a boolean function. The effort to find such a function that corresponds to a given truth table stored in the ROM is proportional to the number of data bits and grows exponentially with the number of address bits.

ProxyPlayerHD, 2019-12-14:
>That's interesting! Which tool chain you are using?

I'm using Quartus II 13.0SP1 if you meant that.

but yea it makes sense, before i had a single ROM for all instructions, it had 17 address lines and 32 data lines. so it was massive... now i have a total of 5 ROMs for the same instructions but each only has 12 address lines and 32 data lines. ((2^12) * 5) is much smaller than (2^17) so it all worked out.

hneemann, 2019-12-15:
You can download and try the pre release build v0.23-179-ga4a3ff5 which you can find [here](https://infdigital.dhbw-mosbach.de/).
It has a new experimental PinControl component which allows to create a bidirectional pin in a HDL:

[image]

But the generated HDL code it is not very well tested and therefore may not work as expected.

ProxyPlayerHD, 2019-12-15:
thanks man, i didn't expect you to jump on that so quickly

I will try to make a simple RAM Interface to test it.

ProxyPlayerHD, 2019-12-15:
Ok here is my result. It appears to be working fine, i was able to Read and Write data from/to the SRAM Chip on my FPGA Dev Board with this circuit:

[image]

only thing i noticed was that you need something after the "RD" pin of the Pin Control so it doesn't constantly read Data, so i used an AND gate that was connected to the "RD-KEY" input so it only actually reads data when i tell it to.

Other than that it appears to be working completely fine.

hneemann, 2019-12-15:
That was fast! Thank you for testing this!

> only thing i noticed was that you need something after the "RD" pin of the Pin Control so it doesn't constantly read Data, so i used an AND gate that was connected to the "RD-KEY" input so it only actually reads data when i tell it to.

I'm not sure I got that.

> lastly, this made me notice the fact that my SRAM chip seems to be filled with Random Data upon starting the FPGA, i thought only DRAM had these kinds of issues,
> or maybe it's residue data from the example program it runs by default or something...?

A DRAM uses capacitors to store bit, which are all discharged pretty quickly. After switching on, I would expect the memory content to look uniform. A SRAM in contrast uses two inverters wired as FlipFlop to store the bit. Here I would expect a random content after switching on. But many other factors could play a role. I don't think you can make a general statement about what happens to RAM when it is powered off.

ProxyPlayerHD, 2019-12-15:
>I'm not sure I got that.

basically the default state of the Pin Control is to act as an Input, right? so it constantly reads in Data even when you don't want it to. like when you write something TO the pin it also reads it in at the same time... so all the AND gate does is block the unwanted data from getting further into the circuit, or in this case reach the LEDs.

> A DRAM uses capacitors to store bit, which are all discharged pretty quickly. After switching on, I would expect the memory content to look uniform. A SRAM in contrast uses two inverters wired as [...]

oh yea i forgot that Flipflops are a bit weird with their start values. Guess my programs just need to include a function to clear out RAM before doing anything else. I could also keep a few values and use them as RNG seed or something.

hneemann, 2019-12-15:
> basically the default state of the Pin Control is to act as an Input, right? so it constantly reads in Data even when you don't want it to. like when you write something TO the pin it also reads it [...]

Yes. This is the equivalent circuit of the PinControl component:

[image]

And because it's so simple, there was no need to add a built-in component like this to Digital.
And even now a built-in component is usefull only, because it makes it easier for the HDL generator to figure out, what the user want's to do.

ProxyPlayerHD, 2019-12-15:
ok so as a final confirmation i put this new component into my CPU Circuit and made a little program to test the RAM.

all the Program does is it takes a value from an Address, put it into a Register connected to some LEDs on my FPGA Board, then Increments the Value at that Address, and finally just repeats the whole thing again.
so i should just be able to see 8 LEDs constantly count up in Binary.

and that is exactly what i saw. (though i had to massively slow down the CPU in order to actually see the LEDs change).
but yea, it seems to work perfectly fine. thanks for your help man!

now the next project is building a UART in Digital... oh boy.

hneemann, 2019-12-15:
Did you use the Verilog or the VHDL export?

ProxyPlayerHD, 2019-12-15:
Verilog. why?

hneemann, 2019-12-15:
Because if the Verilog export works, this does not mean that the VHDL export also works. It's good to know what works and what isn't tested yet.

ProxyPlayerHD, 2019-12-15:
hmm, well i have no idea about VHDL and i cannot get it to Analyse and Synthesise with the error:
"Error (12007): Top-level design entity "RAM-TEST" is undefined"

which is strange as the file is called "RAM-TEST" and in the project settings it also has "RAM-TEST" set as it's top-level entity. so i'M not sure what i'm doing wrong here...

hneemann, 2019-12-15:
The top level entity in the VHDL export is always named "main".

ProxyPlayerHD, 2019-12-15:
oh... well.

ok so i got it working and onto my FPGA, and it worked as well as the Verilog one.

hneemann, 2019-12-15:
Thank you very much for clarifying this.

ProxyPlayerHD, 2019-12-15:
glad to help!

guess this resolves the issue?

hneemann, 2019-12-15:
Merging the feature branch resolves the issue! :smile:

hneemann, 2019-12-15:
Please try the new [pre release](https://infdigital.dhbw-mosbach.de/) built v0.23-182-g98694a9.
I decided to modify the shape to make the circuit more readable:
[image]
Due to the also modified pin positions your circuits have to be modified as well.

ProxyPlayerHD, 2019-12-15:
looks pretty neat. updated my circuit.

i should really do a UART next to make good use of this Computer... if i knew how to build one in a logic simulator. this is the downside of not being good at Verilog.

hneemann, 2019-12-17:
closed in bd8063903571719377acc5f535b2abf0d325f708

closed 2019-12-17 by hneemann