hneemann/Digital #395: Error with tunnel?

obiwanjacobi, 2019-12-13:
I am not sure what I am doing wrong here. A circuit with an input, splitter and tunnels. I complains that the tunnel is not connected.

[image]

hneemann, 2019-12-13:
A splitter works different in Digital compared to a splitter in Logisim.
In Digital a splitter has a direction which means inputs at one side and outputs at the other.
The net "IO" has no output connected to it, which means the state of the net is undefined.
See also #197

obiwanjacobi, 2019-12-13:
Ah, but then the default configuration of the splitter is set to 'merger' - that is what threw me off.

hneemann, 2019-12-13:
That's right! I hadn't even noticed that yet! :smile:

commit 545e2ab5 (2019-12-14): closes #395
closed 2019-12-14 by hneemann