hneemann/Digital #399: Unable to reuse my own ciruit properly inside of another custom circuit

asarra, 2019-12-18:
I designed a circuit which is a D flip flop, but it won't work the same way the D flip flop from the examples directory does.
It has no difference logicwise.

This is my version: https://ibb.co/5hzsW4t
This is the version from the examples/seq. directory: https://ibb.co/5ksgfsw

hneemann, 2019-12-18:
Please ZIP your circuit and attach the ZIP to this issue.

asarra, 2019-12-18:
[shift register.zip](https://github.com/hneemann/Digital/files/3979702/shift.register.zip)

asarra, 2019-12-18:
[shift register.zip](https://github.com/hneemann/Digital/files/3979724/shift.register.zip)

hneemann, 2019-12-18:
To me the simulation looks perfectly fine. 
The reason for the not working shift register seems to be the usage of latches (level triggered flipflops) instead of edge triggered ones.

asarra, 2019-12-19:
Thanks a lot! The edge triggering version worked fine. I have a question about it though.

It seemed to me that the D NAND flip flop from the examples directory didn't have a pulse transition detector or anything else comparable to it in it. How did that work and my version didn't?

Both were level triggered instead of edge triggered. That is what confuses me the most.

hneemann, 2019-12-19:
> It seemed to me that the D NAND flip flop from the examples directory didn't have a pulse transition detector or anything else comparable to it in it. How did that work and my version didn't?
> Both were level triggered instead of edge triggered. That is what confuses me the most.

That surprises me too! I suspect there's been a misunderstanding.
Could you please attach the working implementation which uses a level triggered flip-flop?

closed 2019-12-22 by hneemann