hneemann/Digital #400: stimulus

amryounis, 2019-12-25:
how one can provide tailored stimulus to inputs. 

the stimulus will be a series of ones and zeros.

hneemann, 2019-12-25:
The easiest way to do that, is to add a test case to the circuit.
The most example circuits contain test cases wich can be used as an example for a test case.

amryounis, 2019-12-25:
a test case is ok but the point is that (as far as I am aware of the app) it will run and issue a success/failure message. So I believe the benefit of a stimulus input is clear here. 

 the other work around would be to have a memory+counter(for address)+clock

hneemann, 2019-12-25:
> the other work around would be to have a memory+counter(for address)+clock

This is also a solution. But using a counter and a ROM you have to check yourself if the circuit works as expected or not.
With the help of a test case this check is also performed automatically and reliably.

amryounis, 2019-12-26:
the Deeds project provide a simple solution:

[image]

closed 2020-01-01 by hneemann