hneemann/Digital #402: Weird issue on FPGA but not in Simulator

ProxyPlayerHD, 2019-12-25:
I'm currently working on a UART, the receiving circuit to be exact. it runs perfectly fine in the Simulator but somehow doesn't when i put it onto my FPGA. (i tested it by slowing the clock down masseively and using LEDs to see what's going on inside)

here a screenshot of the circuit:
[image]
and here the file itself: [DROPBOX](https://www.dropbox.com/s/mqi03tuklk7qcaa/UART-RX.dig?dl=0)
now about the actual issue, the Counter in the middle "MAIN" sometimes increments twice in the same rising edge of the clock on the FPGA, but doesn't do that in the Simulator.
but it seems consistent. it always jumps from 1 to 3, and from 9 to 11. the rest inbetween works normally.

I'm not sure why though, and also not sure where else to ask. but i'd rather ask the guy who made it before i completely redesign the whole circuit. maybe i'm just missing something very obvious.

.

lastly, here the function of the circuit in case it's too much spaghetti to understand:
This entire thing is basically a Finite State Machine. MAIN is the current state, and the top and middle counter are basically just there to "wait" a specific amount of time before continuing. the MUX on the right is used to select the thing that is supposed to increment MAIN in the current state. and the DEC on the left is used to enable a speific counter for that state. (the stuff in the top left is just to divide the clock of the FPGA, and the top right is for the circuit this module is supposed to be put inside, so it can also be ignored)

MAIN Increments the first time once a falling edge from the RX-IN Pin is detected, which signals the start bit of the data coming in.
It increments a second time once the top counter reached it's max value (7), this synchronizes the logic with the incoming data.
the next 8 Increments are done everytime the middle counter reaches it's max value (15), this also shifts data into the shift register (off-screen at the bottom)
Now MAIN should've reached 10, which means the Comparator activates the clear signal and resets the whole thing back to 0. ending the receiving of a byte.

amryounis, 2019-12-26:
it looks like the input to the inverter who provides clock to the main counter is driven by two outputs. one from the D DFF and one from the input BAUD-16 CLK

[image]
[image]

ProxyPlayerHD, 2019-12-26:
well that doesn't make much sense, as seen in the screenshot BAUD-16-CLK is not connected, for test reasons. the baud rate is generated internally from the CLK Input because i cannot really have an external baud generator when i test this circuit on it's own

it also isn't connected in the circuit i send... so where did you get that connection from?

hneemann, 2019-12-26:
At the top of the circuit is a D-flipflop which is used as a clock divider. It is clocked by a comparator. This is quite dangerous because the comparator is synthesized to a complex combinatorial circuit. 
If the input "a" of the comparator is incremented, hazards can occur at the output "=", which then would trigger the following flipflop.
In the simulation of the comparator, however, such hazards do not exist.

Please try this hazard free circuit:

[image]

Because of the high bit width of the comparator it is not so easy to check the combination of comparator and the constant 12499999 for hazards. But this is my first guess for the cause of the problem.

ProxyPlayerHD, 2019-12-26:
but the comaprator also needs to reset the counter, otherwise it's not divided by an equal amount as the time it takes from 0 to the constant is shorter (or sometimes longer) than the time it takes from the constant to 0 again (overflow). so it needs to be reset once it reaches that value.

still i will try this and edit this comment with results

hneemann, 2019-12-26:
> maybe there is some issue with that pin in the verilog generation?

Both the Verilog and the VHDL template seem to be ok.
If a comparator solves the problem, could the fact that ofv goes to 0 when en becomes zero causes a problem?
But to be honest, I still think there is some timing problem because many flipflops/counters depend on a clock signal that is not the one generated by the FPGA clock circuit.

ProxyPlayerHD, 2019-12-26:
>could the fact that ofv goes to 0 when en becomes zero causes a problem?

not sure, that would just effect the EN input on the MAIN counter, not the clock itself, so why would it then increment twice just because the EN input is set to 0 in a single clock cycle?

>But to be honest, I still think there is some timing problem because many flipflops/counters depend on a clock signal that is not the one generated by the FPGA clock circuit.

not really sure what you mean. none of the FFs and Counters run off the FPGA clock directly. they are all based on the same divided clock so i don't see how they could somehow de-sync themself  from eachother (if you mean something like that).

plus i'm not sure how else i should divide the clock by some amount.

anyway now since it seems to work i'm testing my circular buffer. it's just a dual port RAM module with one counter for writing data, and one for reading data, the RD-PNT cannot increment when it's equal to the WR-PNT, and the WR-PNT cannot increment when it's equal to the RD-PNT - 1.
When data is received in the UART it writes it to the Buffer and then increments the WR-PNT 2 clock cycles later. when the CPU reads from the Buffer it reads out the value and then the RD-PNT gets incremented 2 clock cycles later.

I used as imple example program to read out data from the UART once it's received, and then just sends it back over the UART to my PC.
problem is that everything i receive back is delayed by 1 character. so if i send `ABCDEF` i recieve back `<0>ABCDE` (the <0> is supposed to represent a 0 byte).

I tested the Buffer on it's own in the simualtor and on the FPGA and it seemed to work. and the CPU's timing shouldn't be a problem for the circuit as well. as the data in the Buffer pointed by the RD-PNT is always being output, and the Increment only happens after 2 full clock cycles (a read of my CPU takes half a clock cyle, from the falling edge which activates the RD line, to the next rising edge which latches the data into the Internal Register of the CPU). So i really don't see any way the CPU could read a 0 byte.
Here is the whole circuit: (i could also provide the whole thing, with Main and all Sub Circuits)
[image]
I'm sorry if this isn't the place to ask for help for this but really i don't know where else i could ask. and i'm kinda just at my limit currently.

hneemann, 2019-12-26:
> not really sure what you mean. none of the FFs and Counters run off the FPGA clock directly. they are all based on the same divided clock so i don't see how they could somehow de-sync themself from eachother (if you mean something like that).

This is a clock divider that divides the input clock by four. The clock input of the main counter is driven by a NOR gate.

[image]

Look at the clock input of the Main counter. Every time the clock input sees a rising edge, the counter should be incremented by one, right? See what happens instead!

ProxyPlayerHD, 2019-12-26:
hmm, i see. but how else can i make a clock divider that wouldn't cause these issues? just use a massive AND gate?

hneemann, 2019-12-27:
That's easy: Simply connect all clock inputs to the same physical FPGA clock signal, and control the components with the enable signal instead. In this case, the hdl synthesis tool is able to utilize the clock distribution hardware available in the FPGA to avoid clock delays and other related issues. But more importantly, a hazard on an enable input does not affect the operation, while a hazard on the clock input may well be a problem, as shown in the example above. 

The circuit then looks as follows and works as expected.  Even though the hazard is still there, but now only affects the enable signal, which does not disturb the operation:

[image]

The D flipflops must then be replaced by registers that have an enable input.

ProxyPlayerHD, 2019-12-27:
oh, not sure if that would work, it would require atleast a lot of rewiring. and some circuits couldn't be done like this, like the CPU itself, since it has no enable pin. but the divider for the SYS-CLK which is what the CPU and everything else runs on is always just the input clock divided by a power of 2 which shouldn't be a problem. since there is no comaparator and such.

this is how i currently made the baud generator circuit:
[image]

also you're missing an AND gate at the input of the enable pin. as you still need to control the enable pin with whatever was connected to it before. and without it it would just always increment, especally if you divide it by more than 2.

I'm going to try to rewrite the TX Circuit as a test since i know that one works perfectly fine.

hneemann, 2019-12-27:
> basically your design only works if you divide by 2.

This is not correct! This design principle works in any case. In your case the only thing you need is a signal that becomes one in every sixth clock cycle for one clock cycle. This signal can then be used as an enable signal.
And if you really need different clock domains, it's better to use the clock generator hardware in the FPGA to generate the different clock signals, rather than using a bunch of logic elements not designed for this purpose.

But of course this is not absolutely necessary. You can also design circuits that ignore this practice. This may even be beneficial from time to time. But then you have to think about the negative consequences that this can have, which is not always easy.

ProxyPlayerHD, 2019-12-27:
>This is not correct! This design principle works in any case. In your case the only thing you need is a signal that becomes one in every sixth clock cycle for one clock cycle. This signal can then be used as an enable signal.

then i misunderstood how your design works, as you can see if you just throw a different divider at it it breaks down...

>And if you really need different clock domains, it's better to use the clock generator hardware in the FPGA to generate the different clock signals, rather than using a bunch of logic elements not designed for this purpose.

I didn't know FPGAs had that... nor do i know how to use it.

hneemann, 2019-12-27:
>I'm sorry, then i misunderstood how your design works, as you can see if you just throw a different divider at it it breaks down...

If you want to build a counter that counts with a frequency divided by 5207 you can build the circuit like this:

[image]

The enable signal of the BAUD counter becomes one every 5207 cycles. So all other clock cycles are ignored. This means the counter counts with the clock frequency divided by 5207.

> I didn't know FPGAs had that... nor do i know how to use it.

The distribution of the clock signal evenly over the whole FPGA die area is a serious issue. 
To solve this problem, special resources are built into an FPGA to distribute the clock. In larger FPGAs sometimes there are several PLLs at different locations on the chip, which are then synchronized to provide a synchronous clock everywhere on the chip.
The normal CLBs used for user logic are not able to do this. If the frequency is low, and the circuit is simple, this is not a big deal. But the more complex the circuit and the higher the frequency, the stronger the effect, and the more advantageous it is to use the clock distribution hardware provided in the chip.

Unfortunately, this works differently with all FPGA manufacturers, and also with the different FPGA series from one manufacturer. Therefore it is difficult to integrate it into Digital, as it would have to be tested separately for each chip. At present this is only supported for the ARTIX 7 series from Xilinx, because this is the chip I am testing everything on.
But the tools provided by the manufacturers usually offer a wizzard dialog to generate the special verilog code that is required to generate the clock signals for the selected chip.

ProxyPlayerHD, 2019-12-27:
>If you want to build a counter that counts with a frequency divided by 5207 you can build the circuit like this: [...] The enable signal of the BAUD counter becomes one every 5207 cycles. So all other clock cycles are ignored. This means the counter counts with the clock frequency divided by 5207.

oh i'm stupid i forgot i still had the JK-FF at the output

>But the tools provided by the manufacturers usually offer a wizzard dialog to generate the special verilog code that is required to generate the clock signals for the selected chip.

yea i saw that. Quartus has a "MegaWizard" for a lot of build in features. one of which can control the global clock on the FPGA (if i understood it correctly)

I just need to see online how to set it up and then i assume i just feed it my divided SYS-CLK

ProxyPlayerHD, 2019-12-27:
wait the fitter already seems to automatically add the SYS-CLK to the global clocks... so do i even need the megafunction?

[image]

hneemann, 2019-12-27:
> wait the fitter already seems to automatically add the SYS-CLK to the global clocks... so do i even need the megafunction?

I'm not sure! But I made a similar observation with the Xilinx tool chain: In the VGA example, I use the clock hardware to generate the correct VGA pixel clock, but then I divide the clock by two with a simple clock divider to derive the processor clock. The Xilinx tool chain detects this and generates a second clock domain without explicitly generating one.

hneemann, 2019-12-27:
> oh i think i just realized what you meant with "integrate it into Digital". of course... i need digital to generate the circuit but because it doesn't support these i would have to edit the generated file everytime...

It is easier to create the main Verilog code yourself, which contains only the clock generation and one instance of the Digital circuit. Then the generated code can live in one file and the clock generation in another, which do not interfere.

ProxyPlayerHD, 2019-12-27:
>I'm not sure! But I made a similar observation with the Xilinx tool chain: In the VGA example, I use the clock hardware to generate the correct VGA pixel clock, but then I divide the clock by two with a simple clock divider to derive the processor clock. The Xilinx tool chain detects this and generates a second clock domain without explicitly declaring one.

guess the Program is smarter than expected, it sees a fast changing signal used in a lot of things and declares it as a global clock.

>It is easier to create the main Verilog code yourself, which contains only the clock generation and one instance of the Digital circuit. Then the generated code can live in one file and the clock generation in another, which do not interfere.

well that would certainly make things easier. why didn't i think of that? anyways doesn't seem like i would need that right now as it already uses global clocks aparently.

hneemann, 2019-12-27:
> guess the Program is smarter than expected, it sees a fast changing signal used in a lot of things and declares it as a global clock.

The problem is: Can you rely on this always working?
I don't think so! It's better to declare the clock domains explicitly!

ProxyPlayerHD, 2019-12-28:
I'm almost tempted to just say "fuck it" and only put the nessesary stuff on the FPGA, and do everything else externally.
so the FPGA would only contain the CPU, Decoder Logic, Clock Divider, some ROM, and possibly a VGA Video Chip if i ever get around to that. everything else (RAM, Serial Interface, SD Card, etc) would be external.
Pros:
- I don't need to design the circuits myself.
- It probably works a lot more reliable and faster since it's actual hardware and not a custom programmed circuit.
- Less Logic in the FPGA = less space used, so the Design can fit onto smaller and cheaper FPGAs
Cons:
- If/when i put this on it's own PCB it requires more ICs, space, and external wiring.

I'm not sure how to proceed...

hneemann, 2019-12-28:
Hmm! One important aspect is missing: flexibility!
Everything that runs on an FPGA can be modified and adapted without any problems. 
There is no need to layout and order a new PCB, no soldering is necessary. A modification is a matter of minutes. This gives you the freedom to try even crazy things. It doesn't take long and costs you nothing. I like this aspect of an FPGA the most.

BTW: The examples folder contains a VGA video card. To save memory, it only supports a 80x25 text mode, but the most complex thing is the sync signal generator, which can be reused in any case.

ProxyPlayerHD, 2019-12-28:
>flexibility

That is true but just look at it! [LINK](https://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT240X.pdf)
This thing handles the USB Protocol, Serial Transfer, and FIFO on it's own. You just connect the Data I/O, RD/WR, and Power lines and you're good to go. no adpater, no RS232, no weird clock divider, etc. what more could you want from a single chip in terms of Serial transfer?

though true it requires more physical effort because you need to solder it, design the PCB, etc.
and since i want to get into stuff like that anyways it seems fitting. though i have no idea how to use KiCad, similar software, or how to even program a standalone FPGA... but that's something for future me to worry about.

about the VGA, the thing is that the Timing circuit is actually the easiest part, since you just need a few counters and comparators to send out the signals at the correct time.
plus i looked very hard into how the NES for example draws it's tile graphics and i already have a pretty good idea on how it's supposed to work and how i want it to work, the problem is just that I'm stuck like on the UART, where i knew how it's supposed to work but just cannot come up with an idea on how to build the circuit itself.

hneemann, 2019-12-28:
Yes, the FTDI chips are really handy. 
But if you are using an FPGA prototype board, I would expect that such a chip is already available.

ProxyPlayerHD, 2019-12-28:
well the DE2 Board does have a UART... but it's a lot more complicated and requires a lot more work to get it to function. [DATASHEET](file:///Z:/Logisim/CPU_Project_65100/Digital/DE2_System_v1.5/Datasheets/USB/ISP1362%20DatsSheet.pdf)
it seems to be designed to actually interface with USB Decives instead of just functioning as a simple Serial Interface.

hneemann, 2019-12-29:
I have attached a simple UART-RX circuit. It may not work! I don't have any hardware at home to test the circuit on an FPGA, but it might be sufficient to give you an idea how it could be done.

[UART-RX.zip](https://github.com/hneemann/Digital/files/4007953/UART-RX.zip)

ProxyPlayerHD, 2019-12-29:
Oh i see how you're doing this. pretty clever to use a ROM to set the next sample point. gets rid of having to reset the counter each time.

though i noticed it uses the rising edge of RX to start reading in data, which is a bit weird because with RS232 the serial lines are by default high. and get set to low when the data starts.
[image]
This should be easy to change though, a simple NOT gate infront of the RS Latch should reverse the edge that it reacts to.

hneemann, 2019-12-29:
> pretty clever to use a ROM to set the next sample point. gets rid of having to reset the counter each time.

It seemed to me the easiest way to deal with the problem that the first sample point is 1.5 bit times after the falling edge of the start bit and the distance to all following sample points is one bit time long.

> though i noticed it uses the rising edge of RX to start reading in data, which is a bit weird because with RS232 the serial lines are by default high. and get set to low when the data starts.
image

You're right! An inverter is missing.

ProxyPlayerHD, 2019-12-30:
ok i removed the Circular buffer circuit so i can directly test the RX Circuit.

it doesn't apprear to work though, I hooked it up like i assumed... EN connected to the BAUD-16 clock and CLK to the SYS-CLK
This was the circuit to make the BAUD-16 signal, it's only active for one SYS-CLK cycle every 5207 cycles
[image]

hneemann, 2020-01-02:
Which clock frequency and baud rate do you use?
If you want to achieve 9600 baud, a clock divider of 5207 requires a clock frequency of 800MHz (9600*16*5207) and if you want to achieve a higher baud rate, a even higher clock frequency is required.

ProxyPlayerHD, 2020-01-02:
I'm using a baud rate of 300 for testing.

ProxyPlayerHD, 2020-02-03:
I think i'll give up on the UART for now, the circuit works perfectly in the simulator but breaks down in the FPGA.

I think i'll just use some FTDI Chip to handle the serial interface, plus it does it for USB instead of RS232 so that's a bonus.

hneemann, 2020-02-03:
I actually planned to implement a UART on one of my boards, but haven't gotten around to it yet.
At the moment I don't have the time.

closed 2021-07-26 by ProxyPlayerHD