hneemann/Digital #403: Component output doesn't work when connected

elek, 2019-12-29:
I have a complex design where one of the registers doesn't work if it's connected to an EEPROM. If it's connected to an output instead of the address pin of the EEPROM it works well.

Reproduce:

1. clone https://github.com/elek/8bit-cpu-digital
2. Open `computer.dig`
3. Adjust the "circuit specific settings" with setting the current path of "memory" file from the repository to the "Program file"
4. Start simulation
5. Click to the clock 11 times (`MCCMD` output should be `F65F` and the clock should be high) 
6. The expected `vcx` output is `1E` and the expected `INSTR` value is `1` but they are `0`.

[image]

**If I repeat the test with removing the wire between the INSTR and the split/merge on the bottom left, it works well:**

[image]

But INSTR stops working when the INSTR pin is connected to the EEPROM.

hneemann, 2019-12-29:
With the rising edge of the clock while writing 0x1E to the INSRT register there is a glitch on the address input of the eeprom. For 6 gate delay times a 0x15 appears at the address input of the eeprom which causes the INSTR register to be cleared because IRCLR becomes low.
This is hard to see because if the circuit has stabelized again, the glitch has already disappeared.
If you start the circuit in single-gate mode, this becomes obvious.

elek, 2019-12-29:
Wow, thank you very much the quick answer, you are right, It was my mistake. I tried to isolate the problem with different way (I even started Digital in Java debug mode) but didn't get it without your hint.

Thanks the help, and thanks for the great tool...

closed 2019-12-29 by elek