hneemann/Digital #404: I have several problems. I don't know if it's my usage problem or if the program has bug

kEviN1H, 2019-12-30:
version: v0.23-187-gc283d25

1,
This is the digital tube driver module.
[image]
I used this digital tube driver module to splice it to simulate,  There was a mistake from right to left. The second "a" of the digital tube was not connected. But I've already connected. Then I delete this connection and reconnect. Run the simulation again,This time, the "e" of the second nixie tube from the right to the left is not connected,This error will always have no connection rotation on these two addresses

[image]

2
I copy and send the processor.dig routine to another address,Open ALU, save another ALU and modify it. It can simulate
[image]
Then call in processor.dig, this module, prompt the error that there is an output interface pin not found
[image]

hneemann, 2019-12-30:
1. Also the splitter has inputs and a outputs. The simple rule is: 
Connect exactly one red dot to as many blue dots as required.
The state of a net that only connects blue dots is undefined.
See also  #395

2. This problem is unusual. Without the dig files I cannot say 
anything about the cause. Please zip the files and attach the zip-file 
to this issue.

kEviN1H, 2019-12-30:
Thank you for reminding me. The first problem is that the splitter is used in the opposite direction. It's OK to use it in some places. If it's used in the opposite direction, it's wrong. It's OK to change it

Thank you very much.

Second, we also find the problem, which is caused by the custom shape.

Thank you very much.

closed 2019-12-30 by kEviN1H