hneemann/Digital #407: The data load mechanism of ROM should be used for RAM as well

AD-530, 2020-01-01:
I'm working on a banked memory subcircuit and would therefore need to split the data on multiple RAM-blocks / data files. This is not possible with the current approach of a global program memory.

Currently there are two distinct mechanisms to load data on program start into RAM & ROM: For ROM the user can pick one separate data file for each ROM block. For RAM however, there's only one global "program memory" file.
I think the ROM approach is much more powerful and should be used for RAM as well.

hneemann, 2020-01-01:
If you are using a split memory architecture, say even bytes in one chip and odd bytes in another, you can mark all the memories involved as program memory. The file is then distributed to all memories when it is read.
To define a program memory order, the memories are sorted according to the label.

hneemann, 2020-01-07:
See also #167

closed 2020-01-07 by hneemann