hneemann/Digital #408: Inputs connecting directly to outputs

KiloBGram, 2020-01-04:
I have cases when I have a large register subcircuit to connect to and I use a splitter to concatenate the wiring. The issue is, it will not let me directly connect the splitter's output with the input of the register. It makes me painstakingly connect a small wire between the splitter  at every connection point. Would it be possible like in logism to have the option to connect inputs to outputs directly?

hneemann, 2020-01-04:
See #274
Maybe I should think about it again.

KiloBGram, 2020-01-04:
Auto wire works some of the time for me. If it is too complicated then I guess I can get used to it. Would it just be possible to create a 0-width wire where two connections overlap?

hneemann, 2020-01-04:
> Would it just be possible to create a 0-width wire where two connections overlap?

No, a zero-length wire is not possible. And that wouldn't make much sense either, because then you could just as well draw a longer wire.

hneemann, 2020-01-05:
@KiloBGram 
When rethinking this, I found a way to implement the direct connection of pins in a more simple way.

Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) build

KiloBGram, 2020-01-05:
Seems pretty good. Thanks a lot.

commit bb4955bc (2020-01-05): overlapping pins are now connected, closes #408
closed 2020-01-05 by KiloBGram