hneemann/Digital #412: Batch simulation

mguthaus, 2020-01-24:
Have you thought about adding a batch simulation capability? Something simple that you can apply inputs and read outputs and state. One very useful feature for large classes is to do batch simulation for grading.

I ask, because this is a nice replacement for MML (MultiMedia Logic) that many introductory courses use. However, MML is not maintained.

hneemann, 2020-01-24:
Please look at issue #353
Does this solution work for you?

mguthaus, 2020-01-24:
Oh, yes, that is exactly what we would need. Is there more documentation or examples on how to use that? I did not see in the help, but maybe I'm not looking in the right place.

hneemann, 2020-01-24:
There is actually nothing else to document except the command line. But you are right: It should be described in the documentation.
The test cases are normal Digita" test cases. There is nothing special about them.

mguthaus, 2020-01-24:
Yes, I found the description of the test cases down in the context menu.
Perhaps that should be in the manual in addition to the command line
options. Thanks for your hard work!

closed 2020-01-25 by hneemann

jyuan2pace, 2020-01-25:
Is this feature currently in the current release/master branch? I saw it is in the release notes of 0.24. Wonder if it has been included in current release as the pre-release feature has expired in #353 Thank you.

hneemann, 2020-01-26:
@jyuan2pace 
It is available in the current master branch and will be included in the next, soon to be released version. In the meantime, you can try the latest pre-release build available [here](https://infdigital.dhbw-mosbach.de/).

jyuan2pace, 2020-01-26:
Thanks. Not sure if it's only me but the pre release link doesn't work for me. I can build from master anyway. Thanks!

获取 Outlook for Android<https://aka.ms/ghei36>

commit d20b3670 (2020-01-25): adds description of script based testing to the documentation, closes #412