hneemann/Digital #413: Rework tunnel placement

elijahlorden, 2020-01-28:
When using the included DIL chips, I find that placing tunnels on the pins rather than merging lines into a bus looks better and is less cumbersome.  However, having to rename many tunnels is a bit time-consuming.  

I use EasyEDA for PCB layout, and that tool also includes tunnels (in EasyEDA they are called 'net ports').  EasyEDA remembers the name of the last port placed or edited when placing a new port, and auto-increments the label if it contains a number.  Adding this functionality to tunnels would decrease the amount of time needed to construct circuits with many tunnels.

hneemann, 2020-01-28:
A similar function already exists when creating several inputs or outputs.
If you name an input or output and the name ends with a number, this number is incremented when the port is duplicated with copy and paste.
This function is deactivated for tunnels, because you often have to copy tunnels unchanged.
Your suggestion might be a good alternative. I have to think about it.

hneemann, 2020-01-28:
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) build.
What do you think about it?

elijahlorden, 2020-01-28:
Wow, that was fast.  That is pretty much exactly what I had in mind!  

One other change I will suggest, though, is that a selected component is kept on the cursor after placement, until it is cleared by a right-click.  This would allow you to place multiples of the same component quickly without having to move the mouse back and forth between the component menu and the diagram.

hneemann, 2020-01-28:
The hotkey "L" inserts the last inserted component again. 
And the hotkey "T" always inserts a new tunnel.

elijahlorden, 2020-01-29:
I didn't realize that, my bad.  This works perfectly, then.

commit d01f1183 (2020-01-29): closes #413
closed 2020-01-29 by hneemann