hneemann/Digital #417: Problems with the VGA component

ProxyPlayerHD, 2020-02-03:
I'm trying to get the timings for the standard 640x480 video mode but the component doesn't seem to like it.

[image]
acording to the error i nailed the timings perfectly. but it didn't recognize the signal for some reason...

using this site as guide: [LINK](http://tinyvga.com/vga-timing/640x480@60Hz)

hneemann, 2020-02-03:
This [file](https://github.com/hneemann/Digital/blob/master/src/main/java/de/neemann/digital/gui/components/graphics/VGA.java) contains the supported video modes.
In a 640x480 video mode, a hSync len of 120 is not a valid duration. Permitted are 40, 56, 64 and 96.
The link you mention describes an hSync pulse length of 96.

ProxyPlayerHD, 2020-02-03:
ooooh i forgot it doesn't count the front/back porch as sync.

damn. I'm an idiot at times.

ProxyPlayerHD, 2020-02-03:
actually i got a follow up question.

i iut working and the window says "640x480x60Hz"
and i setup 2 comparators to draw lines when X is 320 or when Y is 240. meaning i should have a cross over the screen that goes directly through the middle.

problem is it's not in the middle. is it also drawing the front/back porch for some reason? or why could this be?
[image]

hneemann, 2020-02-03:
Hmm! Especially the mode "640x480x60Hz" I tested quite well. So I don't think there is a bug in the VGA component.
I suspect that the sync pulse is slightly shifted.

ProxyPlayerHD, 2020-02-03:
what do you mean shifted?

the order is always the same for Horizontal and Vertical:

```
Front Porch
Image
Back Porch
Sync
```

but you can choose where to start to the loop from, i always start from the "Image" part since that makes timing a bit easier as pixel 0,0 is always 0,0 in the X and Y Counters for the Timing.
so with that the order of my Timings are:

```
Image
Back Porch
Sync
Front Porch
```
[image]

Here is the full circuit. it's somewhat symmetrical from the middle between the 2 counters. so directly ahead of the counters are the Comparators that reset said counters. the 2 above/below that are for the Sync Signal, and the ones even further from the middle are for drawing the cross.

hneemann, 2020-02-03:
Where the sequence starts does not matter. This is automatically detected by the position of the sync impulses. But the output of the image data must have the correct position relative to the sync pulses.

ProxyPlayerHD, 2020-02-03:
yep, but it doesn't really explain why the image would be shifted by so much, i reddit the math and the values for the comparators are correct.

how did you generate the sync pulses for your testing?

hneemann, 2020-02-03:
I use the VGA example in the processor folder, which outputs the same graphics both in the simulator and via an FPGA on a real monitor.
And I built Ben Eater's VGA circuit, which shows the beautiful [bird](https://www.youtube.com/watch?v=uqY3FMuMuRo) also in the simulation.

ProxyPlayerHD, 2020-02-03:
I'll look at it tomorrow, i'm pretty tired and it's late

ProxyPlayerHD, 2020-02-04:
also when i tried to save the circuit i got the error "There are several files with identical file names, which can not be uniquely assigned! Affected are:"
but then it lists pretty much all files i have which are all named differently... and none of them have the same name as i gave the circuit. so what gives?

ProxyPlayerHD, 2020-02-04:
ok the issue went away on it's own so never mind that i guess.

I'm now trying to get the code onto an FPGA but i'm getting an error of some character that is not allowed in Verilog (the dash or "-"). i use these in every name as you might know, and usually Digital uses the escape character "\" whenever something like that is present....

except for ROMs, it doesn't do that when you got some invalid Verilog character in a ROM's label.
I don't know why but it just doesn't.

hneemann, 2020-02-04:
> except for ROMs, it doesn't do that when you got some invalid Verilog character in a ROM's label.
> I don't know why but it just doesn't. and i always forget that and have to change the name afterwards to something Verilog valid

Yes ROM's must be handled in a special way when they are exported to an HDL. Within the ROM HDL templates the label is used directly and not escaped. I will think about how to fix this.

> oh well, do you know a good way of mapping 16 colors onto a 30 bit color depth?

Use a ROM with 4 address bits and 30 data bits. Then you can easily define the palette as a color table inside the ROM.

ProxyPlayerHD, 2020-02-04:
>Yes ROM's must be handled in a special way when they are exported to an HDL. Within the ROM HDL templates the label is used directly and not escaped. I will think about how to fix this.

thanks but if there is no easy way just leave it. i'll just have to remember that it cannot have special characters in it's Label. maybe just add that to the description or something.

>Use a ROM with 4 address bits and 30 data bits. Then you can easily define the palette as a color table inside the ROM.

well damn i forgot to edit that part out. i was already using that, except i used 3 seperate LUTs, one for Red/Green/Blue

at first i thought it didn't work because the colors were screwed up on the screen. turns out i just can't do math and when choosing the value for the half brightness colors (full = 0x3FF) i choose 0x1F... which is not even close to being half.

so yes i was able to fix that, here a pic of what the test screen looks like.
[image]

ProxyPlayerHD, 2020-02-08:
now this is a bit of an oddity where i'm not sure if i should open a seperate issue for.

I expanded the circuit further and such and exported it to Verilog to test it on my FPGA...
it exported perfectly fine but quartus cannot synthesis the code. it gives the following error:

`Error (10171): Verilog HDL syntax error at vga-main.v(1023) near end of file ;  expecting "endmodule"`

I had this before but cannot remember how i got around it... when i add an "endmodule" manually i get a different error:

`Error (12007): Top-level design entity "VGA-MAIN" is undefined`

which is a bit weird as that was the exact file i converted to Verilog and it had the same name.
I can send you the files for testing if you want.

hneemann, 2020-02-09:
Could you please attach the circuit (zipped *.dig files) which reproduce the faulty verilog code?
Then I can look into it!

ProxyPlayerHD, 2020-02-09:
Here you go, it has all files that should matter

[VGA-Project.zip](https://github.com/hneemann/Digital/files/4176665/VGA-Project.zip)

hneemann, 2020-02-09:
The problem is caused by a splitter used as a merger with a single output port where nothing is connected to. See file VGA-READ.dig. I am afraid that there might be more such unusual corner cases that cause problems.
The latest pre release build fixes this one.

ProxyPlayerHD, 2020-02-09:
thanks.

yea there are a lot of splitters used to extend bits or shift bits. which one exatcly did you mean? can you circle it in a picture?
it would be useful if Digital had a bit extender like Logisim, where you can choose between sign extension, filling the extra bits with 0/1, or having an input that sets the extended bits.

hneemann, 2020-02-09:
In the file VGA-READ.dig there are only three splitters with a single output port with nothing connected to the output port. These three splitters caused the problem.

ProxyPlayerHD, 2020-02-09:
ohhh, those from the registers.

i completely forgot about those. i disconnected them and used constants for testing reasons.

ProxyPlayerHD, 2020-02-09:
that was a misclick

closed 2020-02-09 by ProxyPlayerHD