hneemann/Digital #418: VGA component not showing display on Linux?

harwoodr, 2020-02-03:
My first assumption is (almost always) that I'm doing something wrong, but I'm trying out the latest version's VGA component in the attached circuit.  I do not get any errors and simulation runs - but I get no vga window.  OS is Ubuntu 18.04
[image]

hneemann, 2020-02-03:
I run all tests on a Debian machine, and it works very well, so I don't think there is a problem with Ubuntu.
However, since the VGA component has to measure the timings to detect the video mode, two frames must be output completely before the graphic is displayed. Depending on the clock frequency, this can take a little time.
In any case either a graphics window should appear or an error message stating that the video mode could not be detected.

harwoodr, 2020-02-03:
Ah, patience is a virtue.  Thanks.

closed 2020-02-03 by harwoodr