hneemann/Digital #420: rpm attempt and maven plugin error

copart, 2020-02-10:
Hi,

I am trying to create a rpm (spec) in Fedora to package Digital. However, I am not too familiar with Maven.

I was getting errors on the rpmbuild on the same plugin as shown below.

To get the following, I tried to run "mvn install"

`
~/rpmbuild/BUILD/Digital-0.24]$ mvn install                                                                                                                ⇄
[INFO] Scanning for projects...
[INFO] 
[INFO] ---------------------< de.neemann.digital:digital >---------------------
[INFO] Building Digital 1.0-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- maven-checkstyle-plugin:3.1.0:check (validate) @ digital ---
[INFO] Starting audit...
Audit done.
[INFO] 
[INFO] --- git-commit-id-plugin:2.2.1:revision (get-the-git-infos) @ digital ---
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 2.584 s
[INFO] Finished at: 2020-02-09T22:32:37-05:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal pl.project13.maven:git-commit-id-plugin:2.2.1:revision (get-the-git-infos) on project digital: .git directory is not found! Please specify a valid [dotGitDirectory] in your pom.xml -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
`
However, I really want to be able to create an rpm:

hneemann, 2020-02-10:
The Maven configuration attempts to extract the version number from the git repo in order to add the version to the help dialog and the distribution. I expect that you don't get the sources by "git clone" but by downloading the sources as a zip or tar which means the git repo is missing.
In this case you have to provide the correct version number by a command line argument.

Please refer to issue #289 to see how this can be done. And please make sure that the correct version number is always set.

copart, 2020-02-10:
Thank you! I am closing.

closed 2020-02-10 by copart