hneemann/Digital #421: Feature request: color theme

flucas10, 2020-02-11:
Hello

Adding color themes might be a good idea.
For certain people (and me) who do not like or support too bright colors (Like pure white).

Thanks

hneemann, 2020-02-11:
See also #204

Maybe I should think about it again.

mitolegy, 2020-03-20:
> See also #204
> 
> Maybe I should think about it again.

would absolutely love to see a dark theme. looking at a bright white display for a few hours gets tiresome real fast. maybe you could do something like https://simulator.io/

hneemann, 2020-03-20:
The proble is not to create a dark color theme for the circuit itself. The problem is to create a dark color theme also for the menus and dialogs.
Would it be helpful if only the circuit itself would be displayed in a dark scheme and the menus and dialogs would remain as they are?

mitolegy, 2020-03-20:
> The problem is not to create a dark color theme for the circuit itself. The problem is to create a dark color theme also for the menus and dialogs.
> Would it be helpful if only the circuit itself would be displayed in a dark scheme and the menus and dialogs would remain as they are?

notepad ++ doesn't color its UI elements. can't speak for everyone but for me at least. even tho the UI elements are still white the window as a whole becomes a lot more comfortable to look at. I know Digital's UI is a bit bigger then notepad++'s UI but I think it would still have a similar effect.

if you just go with dark gray instead of black the colors you currently have for wires and components could still be nice and clear. meaning that you could maybe get away with just recoloring the background. maybe inverting the color of the black text written by users.

attached an image experimenting with recoloring the background. the components, and text. just to give some ideas about what might look nice. the bg color is the dark theme background color from discord. the new component color is the same as the color of the gray text around components and the blue for the wires is just a less saturated version of the blue from the default bus wires.

[image]

hneemann, 2020-03-21:
Please try the experiment at the latest [pre release](https://infdigital.dhbw-mosbach.de/).

But to be honest: It looks weird to me if the circuit uses a dark theme and the rest of the user interface doesn't.

mitolegy, 2020-03-21:
it does look a little off. but that might just be getting used to the thing. I can remember youtube and discord looked a bit odd the first time I switched over to dark theme too.

just out of curiosity tho. what part of the UI isn't easy to recolor? is it the UI symbols? or the borders and white fill? and are the components in the component tree also transparent? like they are in the circuit diagram?

hneemann, 2020-03-21:
> Just out of curiosity tho. what part of the UI isn't easy to recolor? is it the UI symbols? or the borders and white fill? and are the components in the component tree also transparent? like they are in the circuit diagram?

First of all: I am not a good designer. It is difficult for me to create a coherent, goog looking color scheme. :smiley: 

And there are some cross-platform issues, especially with the Mac. Retina display support on Mac and MS-Surface requires different approaches. And worst of all, it's difficult to create test cases for these things, and I'm very uncomfortable with code that I can't write tests for. I don't even have access to all these different hardware devices to do manual tests.

Currently, most of the UI stuff seems to work fine everywhere. That' why I don't want to touch the UI code.

closed 2020-06-25 by hneemann

DevCharly, 2020-12-17:
If you guys are still looking for a dark Swing L&F,
you could try FlatLaf (https://github.com/JFormDesigner/FlatLaf).
It comes with dark and light themes and scales on HiDPI screens.

commit 55c48953 (2020-06-25): adds color schemes, closes #481, closes #421