hneemann/Digital #422: Support for low-res VGA

lfjewett, 2020-02-25:
I tried to build a Ben Eater VGA circuit from youtube that uses a 10mhz signal to drive a 25% resolution 800x600 screen.  I'm doing this in hopes of modeling a 50% resolution 640x480 for an 8bit computer build. 320x240 effective. I was unable to get this to work with ether my own circuit or the one provided in the examples folder.  While I can drive the circuit from a 12Mhz clock and it seems to work fine it will only display at the full 640x480 resolution.  Is there any chance the VGA mode will work on these slower clocks?

hneemann, 2020-02-25:
Ben Eater's VGA circuit works fine, but you have to trick the VGA monitor with a prescaler:

[BenEaterVGA.zip](https://github.com/hneemann/Digital/files/4248737/BenEaterVGA.zip)

lfjewett, 2020-02-27:
You are exactly right. The above works just fine and I was able to implement the same for 320x240 as well. Thank you for the correction. Closing ticket.

closed 2020-02-27 by lfjewett