hneemann/Digital #423: use IEEE symbols instead of DIP

Left-Hand, 2020-02-27:
They are 74190,74194 and 74181 ,which should be added in it,and there is a little problem in 74181
[image]
    I really love to use this software to build logic gates,espeaically for my 74 processor,but a  problem is still worrying me,that when i use a new chip, i must make a new svg shape for it,it's  so hard to work with dip shape and the useless power pin,in ord to apdpt to some edas like Kicad

[74_gates.zip](https://github.com/hneemann/Digital/files/4260660/74_gates.zip)

  by the way ,the A=B pin in 74181 is Open collector，i made a little change for it
    but the next problem came,that when i want to update my digital,i try to save my lib, so i just copy the files below
[image]
the new components can be used in it,but there is a bug that when i try to open a digital file,it alaways stay the last file,i can only open the file in it

hneemann, 2020-02-27:
If you want to create a new 74xx circuit you don't need to create a custom shape. Simply select the dil shape and give the inputs and outputs the appropriate number. There is a special wizard which helps with the numbering and also the creation of the power pins can be done automatically.
And even if you have selected the dil shape in the circuit settings, you are still able to select an other shape when using the chip in a circuit. 

> the new components can be used in it,but there is a bug that when i try to open a digital file,it alaways stay the last file,i can only open the file in it

I'm sorry, I'm not quite sure what you mean.

But in one of the latest comits i have moved the library folder. Maybe that's the problem you have?
In this case select the correct folder in the settings.

hneemann, 2020-03-04:
Feel free to reopen the issue if necessary.

closed 2020-03-04 by hneemann