hneemann/Digital #426: Allow counters to be used as Program Counters

binarycow, 2020-03-15:
I really like that a `Register` has the "Program Counter" checkbox, to aid in using a detached debugger.  I request that this functionality be added to `Counter with Preset` and `Counter` as well.

----

For my design, I want to use a `Counter with Preset` as a program counter - it more accurately describes the functionality that a PC will normally have.

With a register, I can only store an arbitrary value, and retrieve that value.  With the `Counter with Preset`, I can not only store an arbitrary value, but I can also increment the counter in only one 'tick', and without tying up an ALU simply to add 1.  And, since the `Counter with Preset` also has an increment/decrement toggle, it can also be used for stack pointers.

I have tried to work around this limitation by using both a `Counter with Preset` and a `Register`, but the workarounds are finnicky and ugly.  And, if/when I implement this design it hardware, that is not necessary, since the `74LS163` fulfills the same functions as `Counter with Preset`.

At present, my only **real** option is to move forward with the `Counter with Preset` as my PC, and I will have to figure out the remote debugging stuff later.

hneemann, 2020-03-15:
The option "Program counter" is only required for the use of the remote interface. This means that an assembler IDE must be implemented, which needs to retrieve the current program counter value from the simulator to mark the corresponding instruction.
If writing your own IDE is not the highest priority, go for the "Counter with Preset ".

closed 2020-03-15 by hneemann

hneemann, 2020-03-15:
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/).

binarycow, 2020-03-16:
Thanks!  I did not expect anything at all, especially this quickly!  I appreciate your hard work!

hneemann, 2020-03-16:
Never mind!
Overall, it even seemed strange to me that the built-in counters could not be used as program counters. But I hadn't noticed it yet.

commit a814f678 (2020-03-15): Counters are able to act as program counters via the remote interface, closes #426