hneemann/Digital #427: Support for custom Application Options in external component

ibfelzmann, 2020-03-20:
I implemented a small extension to allow the user to input their own custom application options to the External component.
Currently only for GHDL.

Check my fork in https://github.com/ibfelzmann/Digital . I believe I am ready to submit a Pull Request.

**Background**
After the COVID-19 pandemic, my Logic Circuits laboratory class is going online.
Usually, we use a DE1-SoC board for students to play with their VHDL designs, but this is no longer possible, and I want to have something to "give them a fell" of having the board.

So I reproduced some of the hardware in the board in Digital and used an External component to hold the VHDL code. The problem is that I still want them to synthesize their code on Quartus Prime, so that we can ensure that it is a synthesizeable code and the environment is a accurate-wish representation of the accurate board.

The solution I found was to synthesize the code on Quartus and export the netlist written for simulation, taking advantage of GHDL vendor support scripts to import the altera libraries. But that means I would require to import custom libraries within Digital External components, or basically send GHDL a -P option -- and, in case of this use case, a -fsynopsys.

Below two screenshots of this working. The bin2hex entity converts a 4-bit input signal from SW to its hexadecimal representation in 7-segments displays (the right-most one).

[image]
[image]

I still intend to build a custom "PS/2 keyboard" component (to generate actual PS/2 signals instead of the actual characters) and play with VGA support later in this semester, if all goes well.

hneemann, 2020-03-20:
I like the idea. But it should also work for IVerilog and generic applications.
I have to think about it.

ibfelzmann, 2020-03-21:
I believe the same implementation works for IVerilog as well. The custom options would be inserted just after the -tvvp argument in ApplicationIVerilog::start and ApplicationIVerilog::checkCode.
Only the default value would be different - blank in case of IVerilog.

Actually, the default options for GHDL are --std=08 --ieee=synopsys, but maybe these could be left blank as well, since --ieee=synopsys has been deprecated and leaving it hard-coded as a default option could end up linking Digital to some older release of GHDL.

As for generic, it is already a full custom application that just calls whatever is written in the "code" field, isn't it? Additional custom options do not seem to be applicable in this scenario.

hneemann, 2020-03-22:
Do you expect to need different sets of options in different external components?
I'm thinking of adding the options to the settings.

hneemann, 2020-03-22:
See e33e9a148878fa168b47abe766879273e5c3ecdb

Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/).

hneemann, 2020-03-23:
@ibfelzmann 
When I thought about it, I came to the conclusion that your proposal would work best.
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) (v0.24-22) and let me know how it works at your use case.

ibfelzmann, 2020-03-23:
Yes, that did the trick.
Only it does not allow escaped spaces within options...

hneemann, 2020-03-24:
Pre release v0.24-23 should also support character escaping.

commit 640a53e6 (2020-03-25): adds external app options, closes #427
closed 2020-03-25 by hneemann